wasm-micro-runtime/wasm-micro-runtime #985: sleep in sub thread cause crash in android or linux

xingkaiyu, 2022-01-25:
hi~
I want to call sleep or nanosleep in my code. 
I find that when I call sleep or nanosleep in main thread it is ok but crash in sub thread.
I also try in windows. They work find both in main and sub thread....

By the way, I want to know why usleep not work in wasm app.

I attached my simple code.
[wasm_source.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/7933980/wasm_source.zip)

wenyongh, 2022-01-26:
@xingkaiyu have you updated the source code to latest? We have submitted patch #964 to fix the issue #961, which is similar to the issue you reported.
BTW, the main function of source code you uploaded seems should be like:
```C
int main(int argc, char** argv)
{
    SleepInSub();
    SleepInMain();
    return 0;
}
```
I built the main.cpp with the following command, and it works with latest iwasm:
```bash
/opt/wasi-sdk/bin/clang++ -pthread -O3 \
    -Wl,--shared-memory,--max-memory=196608 \
    -Wl,--allow-undefined,--no-check-features \
    -Wl,--export=__heap_base,--export=__data_end \
    main.cpp -o test.wasm
```

xingkaiyu, 2022-01-28:
@wenyongh 
I am using the 12-30-2021 release with uvwasi.
And I applyed the patch attached (it is from patch #964). But it does not work for me.
Is there any based patches between release 12-30-2021 and patch #964?
[my-patch.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/7958433/my-patch.zip)

xingkaiyu, 2022-01-28:
I have tried with the newest code. It still goto creash....

wenyongh, 2022-01-29:
@xingkaiyu thanks for reporting the issue, the wasi library you used is libc-uvwasi but not libc-wasi, the former requires more native stack space while calling the uvwasi APIs, sometimes the native stack overflow cannot be detected. I uploaded #992 to fix it, please have a try, the patch is WIP, we may need to test more and improve it in the future. Thanks.

xingkaiyu, 2022-02-02:
I have tried  #992 and it works for me.
Hope you can merge it ASAP.

closed 2022-02-02 by xingkaiyu