wasm-micro-runtime/wasm-micro-runtime #986: invokeNative_em64.s error when build samples on Mac m1 platform

jasonleecode, 2022-01-26:
report error when build samples in folder below:
**wasm-micro-runtime/samples/simple**

/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:9:5: error: unknown directive
    .type invokeNative, @function
    ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:19:10: error: unknown token in expression
    push %rbp
         ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:19:10: error: invalid operand
    push %rbp
         ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:20:9: error: unknown token in expression
    mov %rsp, %rbp
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:20:9: error: invalid operand
    mov %rsp, %rbp
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:22:9: error: unknown token in expression
    mov %rdx, %r10
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:22:9: error: invalid operand
    mov %rdx, %r10
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:23:9: error: unknown token in expression
    mov %rsp, %r11
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:23:9: error: invalid operand
    mov %rsp, %r11
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:24:13: error: unknown token in expression
    and $8, %r11
            ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:24:13: error: invalid operand
    and $8, %r11
            ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:25:5: error: unrecognized instruction mnemonic
    je check_stack_succ
    ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:26:5: error: unrecognized instruction mnemonic, did you mean: hint?
    int3
    ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:28:9: error: unknown token in expression
    mov %r10, %r11
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:28:9: error: invalid operand
    mov %r10, %r11
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:29:13: error: unknown token in expression
    and $1, %r11
            ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:29:13: error: invalid operand
    and $1, %r11
            ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:30:13: error: unknown token in expression
    shl $3, %r11
            ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:30:13: error: invalid operand
    shl $3, %r11
            ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:31:9: error: unknown token in expression
    sub %r11, %rsp
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:31:9: error: invalid operand
    sub %r11, %rsp
        ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:33:10: error: unknown token in expression
    movq %rdi, %r11
         ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:33:10: error: invalid operand
[... 62 lines cut ...]
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:56:14: error: unexpected token in argument list
    movq 0x58(%rsi), %rcx
             ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:57:14: error: unexpected token in argument list
    movq 0x60(%rsi), %r8
             ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:58:14: error: unexpected token in argument list
    movq 0x68(%rsi), %r9
             ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:59:14: error: unexpected token in argument list
    movq 0x48(%rsi), %rsi
             ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:61:10: error: unknown token in expression
    call *%r11
         ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:61:10: error: invalid operand
    call *%r11
         ^
/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:62:5: error: unrecognized instruction mnemonic
    leave
    ^
make[2]: *** [CMakeFiles/vmlib.dir/Users/jasonlee/Documents/code/opensource/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s.o] Error 1
make[1]: *** [CMakeFiles/vmlib.dir/all] Error 2
make: *** [all] Error 2
Failed to build runtime sdk

wenyongh, 2022-01-29:
Hi, seems that the build target isn't set correctly, could you please try building them with `cmake .. -DWAMR_BUILD_TARGET=AARCH64`?

jasonleecode, 2022-02-03:
ok, I will check this configuration.

jasonleecode, 2022-02-07:
It still doesn't work after add cmake flag 'DWAMR_BUILD_TARGET=AARCH64' in the build script.

wenyongh, 2022-02-14:
@jasonleecode have you make clean and make again, please check whether invokeNative_em64.s is still being compiled? Normally file invokeNative_aarch64.s should be compiled instead of invokeNative_em64.s for ARM64 target.

jasonleecode, 2022-02-20:
@wenyongh unfortunately i have tested this option, and it doesn't work. maybe m1 silicon platform is too new to support, there is some special environment variables that we don't know yet.

wenyongh, 2022-03-08:
@jasonleecode Did you build iwasm under the folder `product-mini/platforms/darwin` but not `product-mini/platforms/linux`? In MacOS, we should use the former to build iwasm.

Lost-Temple, 2022-08-31:
> build

I builded iwasm under the folder `product-mini/platforms/darwin`  , invokeNative_em64.s error when build `wamr-sdk` on Mac m1 platform

Lost-Temple, 2022-09-01:
@wenyongh

wenyongh, 2022-09-01:
@Lost-Temple Could you please try modifying file `wamr-sdk/wamr_config_default.cmake`, change the platform and target:
```cmake
set (WAMR_BUILD_PLATFORM "darwin")
set (WAMR_BUILD_TARGET AARCH64)
```

Lost-Temple, 2022-09-02:
> @Lost-Temple Could you please try modifying file `wamr-sdk/wamr_config_default.cmake`, change the platform and target:
> 
> ```cmake
> set (WAMR_BUILD_PLATFORM "darwin")
> set (WAMR_BUILD_TARGET AARCH64)
> ```

@wenyongh yes,it worked . thank you very much.

closed 2022-09-27 by wenyongh