wasm-micro-runtime/wasm-micro-runtime #991: compiling with Wcast-align shows a huge number of warnings for THUMB

kfessel, 2022-01-28:
When compiling with Wcast-align and build traged beeing THUMB a huge number of warnings is shown.

RIOT uses Wcast-align by default i switched it of for the libwarm.a generation.

```
core/shared/mem-alloc/ems/ems_gc_internal.h:191:18: warning: cast increases required alignment of target type [-Wcast-align]
  191 |                ? (hmu_normal_node_t *)((uint8 *)node + node->next_offset)
```
<details>
```
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Build Configurations:
     Build as target THUMB
     CMAKE_BUILD_TYPE 
     WAMR Interpreter enabled
     WAMR AOT disabled
     WAMR JIT disabled
     Libc builtin enabled
     Libc WASI disabled
     Fast interpreter disabled
     Multiple modules disabled
     Reference types disabled
-- Configuring done
-- Generating done
-- Build files have been written to: .../examples/wasm/bin/nucleo-f767zi/pkg-build/wamr
make -C .../examples/wasm/bin/nucleo-f767zi/pkg-build/wamr 
Scanning dependencies of target wamr
[  3%] Building C object CMakeFiles/wamr.dir.../wamr/core/shared/platform/riot/riot_platform.c.obj
[  7%] Building C object CMakeFiles/wamr.dir.../wamr/core/shared/platform/riot/riot_thread.c.obj
[ 10%] Building C object CMakeFiles/wamr.dir.../wamr/core/shared/platform/riot/riot_time.c.obj
[ 14%] Building C object CMakeFiles/wamr.dir.../wamr/core/shared/mem-alloc/ems/ems_alloc.c.obj
In file included from .../wamr/core/shared/mem-alloc/ems/ems_alloc.c:6:
.../wamr/core/shared/mem-alloc/ems/ems_gc_internal.h: In function 'get_hmu_normal_node_next':
.../wamr/core/shared/mem-alloc/ems/ems_gc_internal.h:191:18: warning: cast increases required alignment of target type [-Wcast-align]
  191 |                ? (hmu_normal_node_t *)((uint8 *)node + node->next_offset)
      |                  ^
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c: In function 'hmu_set_free_size':
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c:189:7: warning: cast increases required alignment of target type [-Wcast-align]
  189 |     *((uint32 *)((char *)hmu + size) - 1) = size;
      |       ^
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c: In function 'alloc_hmu':
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c:332:24: warning: cast increases required alignment of target type [-Wcast-align]
  332 |                 rest = (hmu_t *)(((char *)p) + size);
      |                        ^
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c:340:24: warning: cast increases required alignment of target type [-Wcast-align]
  340 |                 next = (hmu_t *)((char *)p + size);
      |                        ^
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c:388:20: warning: cast increases required alignment of target type [-Wcast-align]
  388 |             rest = (hmu_t *)((char *)last_tp + size);
      |                    ^
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c:395:20: warning: cast increases required alignment of target type [-Wcast-align]
  395 |             next = (hmu_t *)((char *)last_tp + size);
      |                    ^
In file included from .../wamr/core/shared/mem-alloc/ems/ems_alloc.c:6:
.../wamr/core/shared/mem-alloc/ems/ems_alloc.c: In function 'gc_realloc_vo':
.../wamr/core/shared/mem-alloc/ems/ems_gc_internal.h:117:26: warning: cast increases required alignment of target type [-Wcast-align]
  117 | #define obj_to_hmu(obj) ((hmu_t *)((gc_uint8 *)(obj)-OBJ_PREFIX_SIZE) - 1)
[... 361 lines cut ...]
  167 | #define read_uint32(p) TEMPLATE_READ_VALUE(uint32, p)
      |                        ^~~~~~~~~~~~~~~~~~~
.../wamr/core/iwasm/interpreter/wasm_loader.c:3440:15: note: in expansion of macro 'read_uint32'
 3440 |     version = read_uint32(p);
      |               ^~~~~~~~~~~
[ 96%] Building C object CMakeFiles/wamr.dir.../wamr/core/iwasm/interpreter/wasm_runtime.c.obj
.../wamr/core/iwasm/interpreter/wasm_runtime.c: In function 'memory_instantiate':
.../wamr/core/iwasm/interpreter/wasm_runtime.c:227:14: warning: cast increases required alignment of target type [-Wcast-align]
  227 |             *(uint32 *)global_addr = aux_heap_base;
      |              ^
.../wamr/core/iwasm/interpreter/wasm_runtime.c: In function 'wasm_instantiate':
.../wamr/core/iwasm/interpreter/wasm_runtime.c:1278:22: warning: cast increases required alignment of target type [-Wcast-align]
 1278 |                     *(int32 *)global_data = global->initial_value.i32;
      |                      ^
.../wamr/core/iwasm/interpreter/wasm_runtime.c:1402:30: warning: cast increases required alignment of target type [-Wcast-align]
 1402 |         uint32 *table_data = (uint32 *)table->base_addr;
      |                              ^
.../wamr/core/iwasm/interpreter/wasm_runtime.c: In function 'wasm_call_indirect':
.../wamr/core/iwasm/interpreter/wasm_runtime.c:2264:25: warning: cast increases required alignment of target type [-Wcast-align]
 2264 |     function_indices = ((uint32_t *)table_inst->base_addr)[element_indices];
      |                         ^
[100%] Linking C static library libwamr.a
[100%] Built target wamr
```
</details>

wenyongh, 2022-02-17:
Hi, have you found any issue when using WAMR in Thumb? If not, could you please disable `-Wcast-align` when building libwarm.a? It really takes effort to clear the warnings, which aren't reported by other targets.

kfessel, 2022-02-17:
Thanks for taking a look.
For the moment i removed that FLAG for the compilation of WAMR on RIOT, which uses it throughout most of its build process and allready spotted a number of (security) bugs.

I just wanted to make you aware of this issue. -and me-

Please have this open as a reminder until it is solved