wasm-micro-runtime/wasm-micro-runtime #993: MinGW support?

lucianoiam, 2022-02-01:
Question, not an issue.

I am working on desktop [audio plugins](https://github.com/lucianoiam/hiphop) that run AssemblyScript code for DSP. These plugins are compiled using MinGW on Windows.

Currently Wasmer is used as the engine but find it a little bloated because of Rust and also hit some obscure crashes on Windows. Already spent too much time on failed attempts to fix them. It works great on Linux and Mac though.

WAMR seems more suitable for use in a plugin environment because of its small footprint. Since the plugins rely on the Wasm C API, I have been able to quickly switch to WAMR and the result works great on Mac... however I hit a roadblock when I realized the library is not compiling on MinGW out of the box :/

Went some steps forward but this is starting to exponentially take more time, so it is worth asking:

Will be MinGW ever supported?  has anyone tried it?

lucianoiam, 2022-02-03:
This would make WAMR more FOSS friendly. It currently does not compile out of the box on MinGW but the missing gaps are minimal. Here is a workaround:

1. Make sure the correct CMake package is installed
```
pacman -R cmake
pacman -S mingw-w64-x86_64-cmake
```

2. Comment out this line in `core/shared/platform/windows/platform_internal.h`
```
#define OS_ENABLE_HW_BOUND_CHECK
```

3. Add `lWs2_32` to the DLL link options in the root CMakeLists.txt
```
target_link_libraries (iwasm_shared ${LLVM_AVAILABLE_LIBS} ${UV_A_LIBS} -lm -ldl -lpthread -lWs2_32)
```

4. Implement `timespec_get()` in `core/shared/platform/windows/win_time.c` which seems to be missing from MinGW. See code pasted below (requires Windows 8+)

5. Run CMake with some flags
```
mkdir build && cd build
cmake .. -G"Unix Makefiles" -DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_INVOKE_NATIVE_GENERAL=1 -DWAMR_DISABLE_HW_BOUND_CHECK=1
```

6. Enjoy

win_time.c

```C
/*
 * Copyright (C) 2019 Intel Corporation.  All rights reserved.
 * SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception
 */

#include "platform_api_vmcore.h"

// https://patches.dpdk.org/project/dpdk/patch/20210324005008.24705-2-dmitry.kozliuk@gmail.com/

#define TIME_UTC 1

static inline int
rte_timespec_get(struct timespec *now, int base)
{
    /* 100ns ticks from 1601-01-01 to 1970-01-01 */
    static const uint64_t EPOCH = 116444736000000000ULL;
    static const uint64_t TICKS_PER_SEC = 10000000;
    static const uint64_t NS_PER_TICK = 100;

    FILETIME ft;
    uint64_t ticks;

    if (base != TIME_UTC)
        return 0;

    GetSystemTimePreciseAsFileTime(&ft);
    ticks = ((uint64_t)ft.dwHighDateTime << 32) | ft.dwLowDateTime;
    ticks -= EPOCH;
    now->tv_sec = ticks / TICKS_PER_SEC;
    now->tv_nsec = (ticks - now->tv_sec * TICKS_PER_SEC) * NS_PER_TICK;
    return base;
}

uint64
os_time_get_boot_microsecond()
{
    struct timespec ts;
    //timespec_get(&ts, TIME_UTC);
    rte_timespec_get(&ts, TIME_UTC);

    return ((uint64)ts.tv_sec) * 1000 * 1000 + ((uint64)ts.tv_nsec) / 1000;
}
```

closed 2022-02-03 by lucianoiam

xujuntwt95329, 2022-02-04:
Hi @lucianoiam, sorry for the late reply as we are in holiday.

Many thanks for providing the detailed steps for building wamr with MinGW. Feel free to raise issue if you encounter any problems while using the MinGW build :)

lucianoiam, 2022-02-04:
Sure will do! I closed the issue because honestly I have not read the rules and maybe GitHub is not the place for asking such things.

I would love to see WAMR building on MinGW without special care though. The instructions above produce a working library, however:

`WAMR_BUILD_LIBC_WASI=0 `
This means WASI will not be available?

`WAMR_BUILD_INVOKE_NATIVE_GENERAL=1`
Library will use C version of `invokeNative()` instead of ASM, could mean a performance penalty for some cases. Could not get the ASM to compile but it should be possible with MinGW GCC (I am not an expert on that platform).

`WAMR_DISABLE_HW_BOUND_CHECK=0`
Not sure about this one, maybe related to the exceptions flavor?. Leaving it enabled produces a compilation error on this line
https://github.com/bytecodealliance/wasm-micro-runtime/blob/770ca8c90c8803f934bf681cd29949b7fb5291a3/core/iwasm/aot/aot_loader.c#L1574
Because `UnwindInfoAddress` is undefined

Then regarding code the only provision that has to be made is taking care of the missing `timespec_get()` function. Either use a shim like suggested above (copied and pasted from some other project having the same issue) or writing a new `os_time_get_boot_microsecond()` for MinGW.

How should these discrepancies be treated? CMake or preprocessor conditionals? I would like to prepare a PR but I am just arrived to WAMR and not sure I can provide the cleanest solution.

xujuntwt95329, 2022-02-04:
You are always welcome to raise your issues, we are making the world of WebAssembly better

I agree with you that we should make it easier to build WAMR with MinGW, I'll discuss this with my teammates later.

> WAMR_BUILD_LIBC_WASI=0
This means WASI will not be available?

Yes, if you specify `WAMR_BUILD_LIBC_WASI=0` then WASI is disabled. We have enabled WASI on Windows through `uvwasi`, you can use `WAMR_BUILD_LIBC_UVWASI=1`, refer to [build_wamr.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-libc) for more details

> WAMR_BUILD_INVOKE_NATIVE_GENERAL=1
Library will use C version of invokeNative() instead of ASM, could mean a performance penalty for some cases. Could not get the ASM to compile but it should be possible with MinGW GCC (I am not an expert on that platform).

Well, this is related to call convention. We need to check which call convention is used in MinGW and choose the suitable ASM file

> WAMR_DISABLE_HW_BOUND_CHECK=1
Not sure about this one, maybe related to the exceptions flavor?. Leaving it enabled produces a compilation error on this line

I noticed that the `RUNTIME_FUNCTION` type is provided by Windows, see doc [here](https://docs.microsoft.com/en-us/windows/win32/api/winnt/ns-winnt-runtime_function), not sure if MinGW use the same header file with MSVC

You are welcome to submit PR :), I'll also consider these questions after holiday.

lucianoiam, 2022-02-04:
Thanks for reviewing and providing these valuable details!