wasm-micro-runtime/wasm-micro-runtime #1007: Build error when WAMR_BUILD_JIT=1 (Linux x64)

lucianoiam, 2022-02-14:
AOT and interpreter build fine without JIT (default config). LLVM was built with `build_llvm.py`.
Am I doing something wrong?

```
-- Build Configurations:
     Build as target X86_64
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter disabled
     WAMR AOT enabled
     WAMR LLVM Orc Lazy JIT enabled
     Libc builtin enabled
     Libc WASI disabled
     Fast interpreter disabled
     Multiple modules disabled
     Hardware boundary check disabled
     SIMD enabled
     Reference types disabled
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c: In function ‘wasm_interp_call_func_native’:
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:845:22: error: ‘WASMInterpFrame’ has no member named ‘operand’
  845 |     frame->lp = frame->operand;
      |                      ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:886:34: error: ‘WASMInterpFrame’ has no member named ‘ret_offset’
  886 |         prev_frame->lp[prev_frame->ret_offset] = argv_ret[0];
      |                                  ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:889:34: error: ‘WASMInterpFrame’ has no member named ‘ret_offset’
  889 |         prev_frame->lp[prev_frame->ret_offset] = argv_ret[0];
      |                                  ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:890:34: error: ‘WASMInterpFrame’ has no member named ‘ret_offset’
  890 |         prev_frame->lp[prev_frame->ret_offset + 1] = argv_ret[1];
      |                                  ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c: In function ‘wasm_interp_call_func_bytecode’:
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:1200:40: error: ‘WASMInterpFrame’ has no member named ‘ret_offset’
 1200 |                 ret_offset = prev_frame->ret_offset;
      |                                        ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3576:38: error: ‘WASMInterpFrame’ has no member named ‘operand’
 3576 |             outs_area->lp = outs_area->operand + cur_func->const_cell_num;
      |                                      ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3576:60: error: ‘WASMFunctionInstance’ has no member named ‘const_cell_num’; did you mean ‘ret_cell_num’?
 3576 |             outs_area->lp = outs_area->operand + cur_func->const_cell_num;
      |                                                            ^~~~~~~~~~~~~~
      |                                                            ret_cell_num
In file included from /mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_runtime.h:11,
                 from /mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:8:
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/../common/wasm_runtime_common.h:32:10: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   32 |         *(int64 *)(addr) = (int64)(value); \
      |          ^~~~~~~~~~~~~~~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3581:17: note: in expansion of macro ‘PUT_I64_TO_ADDR’
 3581 |                 PUT_I64_TO_ADDR(
      |                 ^~~~~~~~~~~~~~~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3585:31: error: assignment to expression with array type
 3585 |                 outs_area->lp += 2;
      |                               ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3590:30: error: lvalue required as increment operand
 3590 |                 outs_area->lp++;
      |                              ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3604:18: error: ‘WASMInterpFrame’ has no member named ‘ret_offset’
 3604 |             frame->ret_offset = GET_OFFSET();
      |                  ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3642:48: error: ‘WASMFunctionInstance’ has no member named ‘const_cell_num’; did you mean ‘ret_cell_num’?
 3642 |                            + (uint64)cur_func->const_cell_num
      |                                                ^~~~~~~~~~~~~~
      |                                                ret_cell_num
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3661:22: error: ‘WASMInterpFrame’ has no member named ‘operand’
 3661 |                 frame->operand + cur_wasm_func->const_cell_num;
      |                      ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3661:49: error: ‘WASMFunction’ has no member named ‘const_cell_num’; did you mean ‘ret_cell_num’?
 3661 |                 frame->operand + cur_wasm_func->const_cell_num;
      |                                                 ^~~~~~~~~~~~~~
      |                                                 ret_cell_num
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3664:28: error: ‘WASMInterpFrame’ has no member named ‘operand’
 3664 |             word_copy(frame->operand, (uint32 *)cur_wasm_func->consts,
      |                            ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3664:62: error: ‘WASMFunction’ has no member named ‘consts’
 3664 |             word_copy(frame->operand, (uint32 *)cur_wasm_func->consts,
      |                                                              ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3665:38: error: ‘WASMFunction’ has no member named ‘const_cell_num’; did you mean ‘ret_cell_num’?
 3665 |                       cur_wasm_func->const_cell_num);
      |                                      ^~~~~~~~~~~~~~
      |                                      ret_cell_num
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c: In function ‘wasm_interp_call_wasm’:
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3763:22: error: ‘WASMInterpFrame’ has no member named ‘operand’
 3763 |     frame->lp = frame->operand + 0;
      |                      ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3764:10: error: ‘WASMInterpFrame’ has no member named ‘ret_offset’
 3764 |     frame->ret_offset = 0;
      |          ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3767:28: error: ‘WASMInterpFrame’ has no member named ‘operand’
 3767 |         word_copy(outs_area->operand + function->const_cell_num, argv, argc);
      |                            ^~
/mnt/src_linux/hiphop/vendor/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_fast.c:3767:50: error: ‘WASMFunctionInstance’ has no member named ‘const_cell_num’; did you mean ‘ret_cell_num’?
 3767 |         word_copy(outs_area->operand + function->const_cell_num, argv, argc);
      |                                                  ^~~~~~~~~~~~~~
      |                                                  ret_cell_num
gmake[4]: *** [CMakeFiles/iwasm_shared.dir/build.make:406: CMakeFiles/iwasm_shared.dir/core/iwasm/interpreter/wasm_interp_fast.c.o] Error 1
gmake[3]: *** [CMakeFiles/Makefile2:81: CMakeFiles/iwasm_shared.dir/all] Error 2
gmake[2]: *** [Makefile:130: all] Error 2
make[1]: *** [../../Makefile.plugins.mk:291: ../../vendor/wasm-micro-runtime/build/libvmlib.a] Error 2
make: *** [Makefile:6: examples] Error 2
```

lum1n0us, 2022-02-15:
Is it OK to assume the compile command includes:

```
$ cmake -DWAMR_BUILD_INTERP=0 -DWAMR_BUILD_JIT=1 .. 
```

wenyongh, 2022-02-15:
@lucianoiam Seems that you built WAMR with the command that @lum1n0us provides, in fact the interpreter must be enabled when building JIT, so normally we should not add `-DWAMR_BUILD_INTERP=0` when `-DWAMR_BUILD_JIT=1` is added. We submitted patch #1008 to fix the build error, please try again. Thanks.

lucianoiam, 2022-02-15:
@lum1n0us yes, effectively I had a meaningless combination:  `-DWAMR_BUILD_AOT=1 -DWAMR_BUILD_JIT=1  -DWAMR_BUILD_INTERP=0`

@wenyongh #1008 correctly detects and automatically enables the interpreter to prevent this build error.

Thanks!

closed 2022-02-15 by lucianoiam