wasm-micro-runtime/wasm-micro-runtime #1009: [C API] Magic header not detected with AOT

lucianoiam, 2022-02-15:
`wasm_module_new()` fails with this message. 

I have a precompiled wasm module. It loads fine on Linux/Mac/Windows when building WAMR with INTERP=1 AOT=0. The same binary loads after rebuilding WAMR with INTERP=0 AOT=1 on Mac and Windows but not on Linux. All systems Intel x64.

Probably related to https://github.com/bytecodealliance/wasm-micro-runtime/issues/251 . The module data is being loaded like this:

```C
wasm_byte_vec_t moduleBytes;
wasm_byte_vec_new_uninitialized(&moduleBytes, fileSize);

size_t bytesRead = fread(moduleBytes.data, 1, fileSize, file);
```

Not sure this makes sense but already tried something like this at no avail:

```C
wasm_byte_vec_t moduleBytes;
moduleBytes.size = fileSize;
moduleBytes.data = aligned_alloc(4, fileSIze, false);

size_t bytesRead = fread(moduleBytes.data, 1, fileSize, file);
```

I am using C++11 so there is no `std::aligned_alloc`, grabbed [this implementation](https://gist.github.com/dblalock/255e76195676daa5cbc57b9b36d1c99a) instead.

lucianoiam, 2022-02-15:
Update: same issue on Mac... disregard the platform part, blame my project build script.

lum1n0us, 2022-02-16:
May I ask which file was opened? XXX.wasm or XXX.aot .

lucianoiam, 2022-02-16:
I am using the `asc` AssemblyScript compiler [here](https://www.assemblyscript.org/compiler.html#command-line-options) . There is no option for producing .aot files.

These are somehow beginner questions then:

- Will these binaries run unmodified later in a web browser WASM runtime?

- How to go from .ts to .aot?   the compiler above produces also produces .wat

Thanks

lucianoiam, 2022-02-16:
That was a silly question...

`wamrc` compiles .wasm files into .aot which target a specific CPU architecture, hence won't work in a web browser...

> May I ask which file was opened? XXX.wasm or XXX.aot .

Helpful hint, closing now.

closed 2022-02-16 by lucianoiam