wasm-micro-runtime/wasm-micro-runtime #1022: wamrc SIMD compat check fails on fairly standard hardware

lucianoiam, 2022-02-25:
Getting this on 2019 Intel Mac:

```
Error: SIMD compatibility check failed, try adding --cpu=<cpu> to specify a cpu or adding --disable-simd to disable SIMD
```

Appending `--cpu=skylake` fixes it.

Also `./wamrc --cpu=help` does not really list the available CPUs.

lum1n0us, 2022-03-01:
*SIMD compatibility check failed* means the local machine doesn't fully achieve WAMR requirements on SIMD, which is at least *sse4.1*.  I am not sure which kind of CPU a 2019 Intel Mac is using but you are able to check it with `$cat /proc/cpuinfo `

lucianoiam, 2022-03-01:
Hi, just checked and this computer's CPU is i7-8569U

According to Intel it supports SSE 4.1 and 4.2

https://www.intel.com/content/www/us/en/products/sku/191050/intel-core-i78569u-processor-8m-cache-up-to-4-70-ghz/specifications.html

wenyongh, 2022-03-01:
@lucianoiam What is your complete command to run wamrc? In MacOS, the cpu name will be obtained when target/cpu/cpu-features aren't set, could you please help check whether the following code is executed?
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/compilation/aot_llvm.c#L1714

lucianoiam, 2022-03-02:
The complete command is `wamrc --target=x86_64 -o dsp/build/x86_64.aot`

Line 1714 is not being called when using the line above, *but* removing `--target` causes 1714 to be called *and* also the issue goes away (ie, no need to append `--cpu` anymore).

So it seems `--target` and `--cpu` must be both present or absent?

wenyongh, 2022-03-08:
Yes, no need to add `--target` if the target is same as host.

closed 2022-09-24 by wenyongh