wasm-micro-runtime/wasm-micro-runtime #1026: An experimental shared-everything multi module feature enabled at branch dev/link

jhe33, 2022-03-01:
Hi, I commited an experimental shared-everything multi module linking feature recently at dev/link, which based on [dynamic linking conventiion](https://github.com/WebAssembly/tool-conventions/blob/main/DynamicLinking.md). But because the convention was updated during developing, the feature actually implemented the previous version(before July, last year). I hope it is helpful for reducing the compile footprint and time, even in the network env, it helps to reduce the download size.

in the design, there are 2 kinds of modules, root module and dependency module. root module can load only explicitly the dependency modules, and dependency module can load explicitly and implicitly its dependency modules. root modules could be a regular module which has no internal libc implementation, wasi module which has own libc, or AssemblyScript module which has own runtime. dependency module has no own memory allocator, it gets the allocator from root module or wamr builtin. that means all modules will share same linear memory.

explicit loading means explicit dlopen/dlsym/dlclose call are needed, and implicit loading relies on the dylink section in the wasm module. user can get the latter and expose the necessary functions by clang. currently only supports lazy binding, that means all function resolving and linking will occur at invoke time, not at module loading time. so far wasm calling wasm, aot calling aot are well supported. And although the implementation is  platform independent, however I only tested it in linux with limited cases (zlib, cjson etc), I believe there still some situations not covered. I will give several sample code soon.

jhe33, 2022-03-10:
###Table of Contents

[TOC]

- how to build a shared module.
- case 1: regular wasm module links against shared module.
- case 2: wasi module links against shared module.
- case 3: AssemblyScript module links against shared module.
- case 4: explicit link and implicit link

###how to build a shared module.

here we build a cJSON shared module using emsdk. you can use other clang toolchain.

`emsdk/upstream/bin/clang -O3 --target=wasm32-unknown-emscripten -nostdlib -I/usr/include/ -Wl,--shared -fPIC -Wl,--no-entry -Wl,--allow-undefined -Wl,--export=cJSON_Parse -Wl,--export=cJSON_Delete -Wl,--export=cJSON_GetObjectItem -Wl,--export=cJSON_IsString -Wl,--export=cJSON_GetStringValue -Wl,--export=__wasm_call_ctors cJSON.c -o cJSON_shared.wasm`

####tips:
`-Wl,--shared –fPIC` : build PIC wasm module
`--target=wasm32-unknown-emscripten` : emcc dylink supports
`-Wl,--export` : export functions, suggest exposing necessary functions to lighten the link cost.

###case 1: regular module

####build root module
`emsdk/upstream/bin/clang –O3 -Wl,--features=mutable-globals --target=wasm32-unknown-emscripten -nostdlib -I/usr/include/  -Wl,--no-entry -Wl,--export=main -Wl,--export=malloc -Wl,--export=free -Wl,--export=realloc -Wl,--export=__stack_pointer -Wl,--allow-undefined test_parser_shared.c -o test_parser_shared.wasm`

#####tips:
`-Wl,--features=mutable-globals` and `–Wl,--export=_stack_pointer` : expose the __stack_pointer to shared module.
`-Wl,--export` : exposing mem allocator functions is not necessary, which depends on where mem allocator comes from, builtin libc or root module.

####code in root module
declare dlopen/dlsym/dlclose in root module, you can either include dlfcn.h or the following declaration in llvm environment.
```c
__attribute__((import_module("env"), import_name("dlopen"))) void * dlopen(const char *, int);
__attribute__((import_module("env"), import_name("dlsym"))) void * dlsym(void*, const char *);
__attribute__((import_module("env"), import_name("dlclose"))) int dlclose(void *);
__attribute__((import_module("env"), import_name("dltest"))) int dltest(void*);
```

declare function pointers will be returned from dlsym.
```c
typedef cJSON *(*func_parse)(const char *);
typedef cJSON*(*func_getobject)(cJSON *, const char*);
typedef bool(*func_isstring)(cJSON*);
typedef const char*(*func_getstring)(cJSON*);
typedef void(*func_delete)(cJSON*);
```

open a wasm/aot module, currently only supports to pass an absolute path or a file name in the current folder.
```c
void * handle = dlopen("cJSON_shared.aot", 0);
if (!handle) {
	printf("open wasm failed\n");
	return -1;
}
```

get function pointer by dlsym, like regular C function do it.
```c
func_parse parse = dlsym(handle, "cJSON_Parse");
if (!parse) {
	printf("dlsym cJSON_Parse failed\n");
	dlclose(handle);
	return -1;
}
```

call it
```c
cJSON * cjson_handle = parse(json_txt);
[... 177 lines cut ...]
`

then callee

`
emsdk/upstream/bin/clang –O3 --target=wasm32-unknown-emscripten -nostdlib -I/usr/include/ -Wl,--shared -fPIC  -Wl,--no-entry -Wl,--export=print_callee -Wl,--allow-undefined  callee.c –o  callee.wasm callee2.wasm
`

** note: callee2.wasm is appended in the above command. it's like gcc link option "-l", indicates callee implicitly depends on callee2.**

##### tips:
An entry will be added in dylink section of callee module, which indicates callee's dependency.

#### implicit link

we can reference previous cases to do explicit loading, here let’s focus on how callee module implicitly open callee2 module. the following declaration could be applied in LLVM to indicate the import module, import function name and signature, like header file.

callee.c :
```c
__attribute__((import_module("callee2.wasm"), import_name("print_callee2"))) void print_callee2(int);
```

** under the hood, callee.wasm will include an entry in import section. Based on these info (dylink and import sections), VM could implement a link (module loading and function resolving) when the call happens (lazy binding). **

### end

jhe33, 2022-03-11:
[WAMR多模块.pdf](https://github.com/bytecodealliance/wasm-micro-runtime/files/8231343/WAMR.pdf)

andrewrch, 2022-11-12:
Hi @jhe33 @wenyongh 

I'm very interested in this feature. Is it likely this or something similar would find it's way to the main branch? What would be the best way to take this forward, or are there other options being considered for dynamic loading?

My use case is similar to that described - I'd like to download a very small wasm module which can then download and dynamically load more modules depending on when they are required. (E.g. similar to import() in JS).

jhe33, 2022-11-14:
@andrewrch thank you for your interest. Have you tried this branch? from your description, I think it's helpful for your cases.

this patch introduced some changes, like string interning(main has no string interning imeplentation right now), optimized native call (as I know, main has already optimised the native call) etc, not acceptable by main branch. therefore, it's not the proper time merging back to main stream. of cource, rebasing it to main is an alternative choice.

if your case is not sensitive to performance, you can  leverage JS glue code or Host your VM embedded in to workround the calling to other wasm modules.