wasm-micro-runtime/wasm-micro-runtime #1029: Using App-Framework APIs without App-Mgr

egreen64, 2022-03-07:
In looking at the code it looks like the application framework APIs - req/resp, pub/sub, etc. are tied to the application manager. Being that I have already a framework  in place on an MCU to receive/update WASM apps from the cloud, is there any way to leverage the application framework apis without requiring the app manager? What I would like to avoid is the requirement for having a socket connection to a host and to just have the already deployed WASM apps use the application APIs.

xwang98, 2022-03-08:
@egreen64 app-mgr is needed for supporting the messaging and queue among Wasm apps inter-communication. It also interfaces the remote clients through socket, serial, or any other physical communications.

If you don't need app-mgr to receive Wasm app from the link, you can refer to:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/55ad4c7ec73da7f90d697d0d6ce67f083571b121/core/app-mgr/app-manager/app_manager_host.c#L167

If you don't want app-mgr to handle any external communications, just remove the function aee_host_msg_callback or don't call it.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/55ad4c7ec73da7f90d697d0d6ce67f083571b121/core/app-mgr/app-manager/app_manager_host.c#L221

eric-nubix, 2022-04-03:
Great. Thanks for your prompt reaponsd

closed 2022-09-24 by wenyongh