wasm-micro-runtime/wasm-micro-runtime #1031: CMake + WASI-SDK generated an (almost) empty WASM binary?

huming2207, 2022-03-08:
Hi there,

Firstly thanks for your great project!

I'm trying to run WAMR on ESP32-S3. Then I encounter an issue when I started working on the WASM-side project. Basically I've done these steps:

1. I've followed the tutorial here and set up the WASI-SDK and CMake: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-a-project-with-cmake
2. I also cloned this project to my home directory on my host machine
3. I ran:

```
mkdir build
cmake .. -DWASI_SDK_DIR=/opt/wasi-sdk -DCMAKE_TOOLCHAIN_FILE=/home/hu/wasm-micro-runtime/wamr-sdk/app/wamr_toolchain.cmake -DCMAKE_SYSROOT=/home/hu/wasm-micro-runtime/wamr-sdk/app/libc-builtin-sysroot
make
```

Then I've got a 25 byte WASM binary, which is definitely way too small. Here's what it was decompiled into:

```wat
(module
  (memory $memory 1)
  (export "memory" (memory 0)))
```

...and here's my minimal reproducible project:

main.c:

```c
#include <stdio.h>
#include "pthread.h"

void *thread_routine(void *arg)
{
    printf("Enter thread\n");
    pthread_exit(NULL);
    return NULL;
}

int main(int argc, char** argv)
{
    pthread_t tid;

    if (0 != pthread_create(&tid, NULL, thread_routine, NULL)) {
        printf("Failed to create thread\n");
    }

    if (0 != pthread_join(tid, NULL)) {
        printf("Failed to join thread %d.\n", tid);
    }

    printf("Exit\n");

    return 0;
}
```

CMakeLists.txt:

```cmake
cmake_minimum_required(VERSION 3.16)
project(lokmachau-wasm)

add_executable(lokmachau.wasm main.c)
```

May I ask have I missed anything? 

Regards,
Jackson

huming2207, 2022-03-08:
Hi @xujuntwt95329 

Thanks for your reply. Indeed I tried adding something like `--export=main` or `--export=_main` to the linker flags earlier but didn't work.

I've also tried your `set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS},--shared-memory,--max-memory=196608,--export=__heap_base,--export=__data_end")` but it ended up like this:

```
[1/1] Linking C executable lokmachau.wasm
FAILED: lokmachau.wasm 
: && /opt/wasi-sdk/bin/clang --target=wasm32 --sysroot=/home/hu/wasm-micro-runtime/wamr-sdk/app/libc-builtin-sysroot -nostdlib -z stack-size=4096 -pthread -O3 -DNDEBUG -Wl,--initial-memory=65536,--no-entry,--strip-all,--allow-undefined-file=/home/hu/wasm-micro-runtime/wamr-sdk/app/libc-builtin-sysroot/share/defined-symbols.txt,--export=__main_argc_argv,--shared-memory,--max-memory=196608,--export=__heap_base,--export=__data_end CMakeFiles/lokmachau.wasm.dir/main.c.o -o lokmachau.wasm   && :
clang-13: warning: argument unused during compilation: '-pthread' [-Wunused-command-line-argument]
[wasm-validator error in function 0] unexpected false: Atomic operation (atomics are disabled), on 
(i32.atomic.rmw.cmpxchg
 (i32.const 1096)
 (i32.const 0)
 (i32.const 1)
)
[wasm-validator error in function 0] unexpected false: Atomic operation (atomics are disabled), on 
(memory.atomic.wait32
 (i32.const 1096)
 (i32.const 1)
 (i64.const -1)
)
[wasm-validator error in function 0] unexpected false: Bulk memory operation (bulk memory is disabled), on 
(memory.init 0
 (i32.const 1024)
 (i32.const 0)
 (i32.const 69)
)
[wasm-validator error in function 0] unexpected false: Atomic operation (atomics are disabled), on 
(i32.atomic.store
 (i32.const 1096)
 (i32.const 2)
)
[wasm-validator error in function 0] unexpected false: Atomic operation (atomics are disabled), on 
(memory.atomic.notify
 (i32.const 1096)
 (i32.const -1)
)
[wasm-validator error in function 0] unexpected false: Bulk memory operation (bulk memory is disabled), on 
(data.drop 0)
[wasm-validator error in module] unexpected false: memory is shared, but atomics are disabled, on 
memory
[wasm-validator error in module] unexpected false: nonzero segment flags (bulk memory is disabled), on 
Fatal: error validating input
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

``` 

However, if I remove the `--shared-memory,--max-memory=196608`, it successfully compiles with a (seem to be) correct WASM binary generated. Of course it's not with linear memory any more as I disabled it:

```wat
(module
  (type $t0 (func (param i32 i32) (result i32)))
  (type $t1 (func (param i32) (result i32)))
  (type $t2 (func (param i32)))
  (type $t3 (func (param i32 i32 i32 i32) (result i32)))
  (import "env" "puts" (func $env.puts (type $t1)))
  (import "env" "pthread_exit" (func $env.pthread_exit (type $t2)))
  (import "env" "pthread_create" (func $env.pthread_create (type $t3)))
  (import "env" "pthread_join" (func $env.pthread_join (type $t0)))
  (import "env" "printf" (func $env.printf (type $t0)))
  (func $f5 (type $t1) (param $p0 i32) (result i32)
    i32.const 1029
    call $env.puts
    drop
    i32.const 0
    call $env.pthread_exit
    i32.const 0)
  (func $__main_argc_argv (type $t0) (param $p0 i32) (param $p1 i32) (result i32)
    global.get $g0
    i32.const 16
    i32.sub
    local.tee $p0
    global.set $g0
    local.get $p0
    i32.const 12
    i32.add
    i32.const 0
    i32.const 1
    i32.const 0
    call $env.pthread_create
    if $I0
      i32.const 1042
      call $env.puts
      drop
    end
    local.get $p0
    i32.load offset=12
    i32.const 0
    call $env.pthread_join
    if $I1
      local.get $p0
      local.get $p0
      i32.load offset=12
      i32.store
      i32.const 1066
      local.get $p0
      call $env.printf
      drop
    end
    i32.const 1024
    call $env.puts
    drop
    local.get $p0
    i32.const 16
    i32.add
    global.set $g0
    i32.const 0)
  (memory $memory 1)
  (global $g0 (mut i32) (i32.const 5200))
  (global $__data_end i32 (i32.const 1093))
  (global $__heap_base i32 (i32.const 5200))
  (export "memory" (memory 0))
  (export "__main_argc_argv" (func $__main_argc_argv))
  (export "__data_end" (global 1))
  (export "__heap_base" (global 2))
  (data $d0 (i32.const 1024) "Exit\00Enter thread\00Failed to create thread\00Failed to join thread %d.\0a"))
```

I haven't got a chance to put it on ESP32 to run. But it seems legit.

huming2207, 2022-03-08:
Btw I forgot to mention, the WASI-SDK compiler version I've used is v14 with Clang 13:

```
clang version 13.0.0 (https://github.com/llvm/llvm-project fd1d8c2f04dde23bee0fb3a7d069a9b1046da979)
Target: wasm32-unknown-wasi
Thread model: posix
InstalledDir: /opt/wasi-sdk/bin
```

I don't know why the compiler was complaining about the atomic operation as I couldn't find where and why it's disabled. I've also found that Emscripten seems to have a thing called `--no-check-features` to bypass this feature check: https://github.com/emscripten-core/emscripten/issues/8503

I've tried that but didn't work on my WASI-SDK setup.

huming2207, 2022-03-08:
And btw, I'm compiling my WASM project using WAMR's built-in Libc. Is that the cause of this atomic disabled error?

lum1n0us, 2022-03-09:
Here is my CMakeLists.txt

```
cmake_minimum_required(VERSION 3.16)
project(lokmachau-wasm)

add_executable(lokmachau.wasm main.c)
target_compile_options(lokmachau.wasm PRIVATE -pthread)
target_link_options(lokmachau.wasm PRIVATE
  LINKER:--export=__heap_base
  LINKER:--export=__data_end
  LINKER:--export=__main_argc_argv
  LINKER:--shared-memory,--max-memory=196608
  LINKER:--no-check-features
)
```
after compiling with cmake

```
$ cmake -DWASI_SDK_DIR=/opt/wasi-sdk -DCMAKE_TOOLCHAIN_FILE=../../../wamr-sdk/app/wamr_toolchain.cmake -DCMAKE_SYSROOT=../../../wamr-sdk/app/libc-builtin-sysroot/ ..
$ make
```

this is what I got:

```
$ /opt/wabt/bin/wasm-objdump -dx lokmachau.wasm

lokmachau.wasm: file format wasm 0x1

Section Details:

Type[5]:
 - type[0] (i32, i32) -> i32
 - type[1] (i32) -> nil
 - type[2] (i32, i32, i32, i32) -> i32
 - type[3] () -> nil
 - type[4] (i32) -> i32
Import[4]:
 - func[0] sig=0 <env.printf> <- env.printf
 - func[1] sig=1 <env.pthread_exit> <- env.pthread_exit
 - func[2] sig=2 <env.pthread_create> <- env.pthread_create
 - func[3] sig=0 <env.pthread_join> <- env.pthread_join
Function[3]:
 - func[4] sig=3
 - func[5] sig=4
 - func[6] sig=0 <__main_argc_argv>
Table[1]:
 - table[0] type=funcref initial=2 max=2
Memory[1]:
 - memory[0] pages: initial=1 max=3 shared
Global[3]:
 - global[0] i32 mutable=1 - init i32=5200
 - global[1] i32 mutable=0 <__data_end> - init i32=1100
 - global[2] i32 mutable=0 <__heap_base> - init i32=5200
 - Export[4]:
 - memory[0] -> "memory"
 - func[6] <__main_argc_argv> -> "__main_argc_argv"
 - global[1] -> "__data_end"
 - global[2] -> "__heap_base"
Start:
 - start function: 4
Elem[1]:
 - segment[0] flags=0 table=0 count=1 - init i32=1
  - elem[1] = func[5]
DataCount:
 - data count: 1
Code[3]:
 - func[4] size=63
 - func[5] size=99
 - func[6] size=303 <__main_argc_argv>
Data[1]:
 - segment[0] passive size=72
  - 0000000: 4578 6974 0a00 456e 7465 7220 7468 7265  Exit..Enter thre
  - 0000010: 6164 0a00 4661 696c 6564 2074 6f20 6372  ad..Failed to cr
  - 0000020: 6561 7465 2074 6872 6561 640a 0046 6169  eate thread..Fai
  - 0000030: 6c65 6420 746f 206a 6f69 6e20 7468 7265  led to join thre
  - 0000040: 6164 2025 642e 0a00                      ad %d...

Code Disassembly:

0000e6 func[4]:
 0000e7: 41 c8 08                   | i32.const 1096
 0000ea: 41 00                      | i32.const 0
 0000ec: 41 01                      | i32.const 1
 0000ee: fe 48 02 00                | i32.atomic.rmw.cmpxchg 2 0
 0000f2: 04 40                      | if
 0000f4: 41 c8 08                   |   i32.const 1096
 0000f7: 41 01                      |   i32.const 1
 0000f9: 42 7f                      |   i64.const -1
 0000fb: fe 01 02 00                |   memory.atomic.wait32 2 0
 0000ff: 1a                         |   drop
 000100: 05                         | else
 000101: 41 80 08                   |   i32.const 1024
 000104: 41 00                      |   i32.const 0
 000106: 41 c8 00                   |   i32.const 72
 000109: fc 08 00 00                |   memory.init 0 0
 00010d: 41 c8 08                   |   i32.const 1096
 000110: 41 02                      |   i32.const 2
 000112: fe 17 02 00                |   i32.atomic.store 2 0
 000116: 41 c8 08                   |   i32.const 1096
 000119: 41 7f                      |   i32.const 4294967295
 00011b: fe 00 02 00                |   memory.atomic.notify 2 0
 00011f: 1a                         |   drop
 000120: 0b                         | end
 000121: fc 09 00                   | data.drop 0
 000124: 0b                         | end
 000124: 0b                         | end                                                                                                                                          [127/576]000126 func[5]:
 000127: 09 7f                      | local[0..8] type=i32
 000129: 23 80 80 80 80 00          | global.get 0
 00012f: 21 01                      | local.set 1
 000131: 41 10                      | i32.const 16
 000133: 21 02                      | local.set 2
 000135: 20 01                      | local.get 1
 000137: 20 02                      | local.get 2
 000139: 6b                         | i32.sub
 00013a: 21 03                      | local.set 3
 00013c: 20 03                      | local.get 3
 00013e: 24 80 80 80 80 00          | global.set 0
 000144: 20 03                      | local.get 3
 000146: 20 00                      | local.get 0
 000148: 36 02 0c                   | i32.store 2 12
 00014b: 41 86 88 80 80 00          | i32.const 1030
 000151: 21 04                      | local.set 4
 000153: 41 00                      | i32.const 0
 000155: 21 05                      | local.set 5
 000157: 20 04                      | local.get 4
 000159: 20 05                      | local.get 5
 00015b: 10 80 80 80 80 00          | call 0 <env.printf>
 000161: 1a                         | drop
 000162: 41 00                      | i32.const 0
 000164: 21 06                      | local.set 6
 000166: 20 06                      | local.get 6
 000168: 10 81 80 80 80 00          | call 1 <env.pthread_exit>
 00016e: 41 00                      | i32.const 0
 000170: 21 07                      | local.set 7
 000172: 41 10                      | i32.const 16
 000174: 21 08                      | local.set 8
 000176: 20 03                      | local.get 3
 000178: 20 08                      | local.get 8
 00017a: 6a                         | i32.add
 00017b: 21 09                      | local.set 9
 00017d: 20 09                      | local.get 9
 00017f: 24 80 80 80 80 00          | global.set 0
 000185: 20 07                      | local.get 7
 000187: 0f                         | return
 000188: 0b                         | end
00018b func[6] <__main_argc_argv>:
  ...
```

huming2207, 2022-03-09:
Thanks @lum1n0us 

> ```
> target_link_options(lokmachau.wasm PRIVATE
>   LINKER:--export=__heap_base
>   LINKER:--export=__data_end
>   LINKER:--export=__main_argc_argv
>   LINKER:--shared-memory,--max-memory=196608
>   LINKER:--no-check-features
> )
> ```

Hmmm, this should be the same as the one I used earlier. I will have another try with this tonight.

xujuntwt95329, 2022-03-09:
> Thanks @lum1n0us
> target_link_options(lokmachau.wasm PRIVATE
  LINKER:--export=__heap_base
  LINKER:--export=__data_end
  LINKER:--export=__main_argc_argv
  LINKER:--shared-memory,--max-memory=196608
  LINKER:--no-check-features
)
> Hmmm, this should be the same as the one I used earlier. I will have another try with this tonight.

@huming2207 I tested this with `wasi-sdk-14` and it works. The complain about atomic operations can be avoid by `--no-check-features`

> Indeed I tried adding something like --export=main or --export=_main to the linker flags earlier but didn't work.

Since `wasi-sdk-12`, the `main` function with arguments is treated as `__main_argc_argv`

lum1n0us, 2022-03-09:
another way will be:

``` c
__attribute__((export_name("_start")))
int
main(int argc, char **argv)
```

huming2207, 2022-03-09:
Hi @lum1n0us @xujuntwt95329 

Looks like the `atomic disabled` issue is related to CMake with Release mode. In Debug mode it works just fine and generated this binary:

```wat
(module
  (type $t0 (func (param i32 i32) (result i32)))
  (type $t1 (func (param i32)))
  (type $t2 (func (param i32 i32 i32 i32) (result i32)))
  (type $t3 (func))
  (type $t4 (func (param i32) (result i32)))
  (import "env" "printf" (func $env.printf (type $t0)))
  (import "env" "pthread_exit" (func $env.pthread_exit (type $t1)))
  (import "env" "pthread_create" (func $env.pthread_create (type $t2)))
  (import "env" "pthread_join" (func $env.pthread_join (type $t0)))
  (func $f4 (type $t3)
    i32.const 1096
    i32.const 0
    i32.const 1
    i32.atomic.rmw.cmpxchg
    if $I0
      i32.const 1096
      i32.const 1
      i64.const -1
      i32.atomic.wait
      drop
    else
      i32.const 1024
      i32.const 0
      i32.const 72
      memory.init $d0
      i32.const 1096
      i32.const 2
      i32.atomic.store
      i32.const 1096
      i32.const -1
      atomic.notify
      drop
    end
    data.drop $d0)
  (func $f5 (type $t4) (param $p0 i32) (result i32)
    (local $l1 i32) (local $l2 i32) (local $l3 i32) (local $l4 i32) (local $l5 i32) (local $l6 i32) (local $l7 i32) (local $l8 i32) (local $l9 i32)
    global.get $g0
    local.set $l1
    i32.const 16
    local.set $l2
    local.get $l1
    local.get $l2
    i32.sub
    local.set $l3
    local.get $l3
    global.set $g0
    local.get $l3
    local.get $p0
    i32.store offset=12
    i32.const 1030
    local.set $l4
    i32.const 0
    local.set $l5
    local.get $l4
    local.get $l5
    call $env.printf
    drop
    i32.const 0
    local.set $l6
    local.get $l6
    call $env.pthread_exit
    i32.const 0
    local.set $l7
    i32.const 16
[... 137 lines cut ...]
    local.get $l35
    global.set $g0
    local.get $l33
    return)
  (table $T0 2 2 funcref)
  (memory $memory 1 3 shared)
  (global $g0 (mut i32) (i32.const 5200))
  (global $__data_end i32 (i32.const 1100))
  (global $__heap_base i32 (i32.const 5200))
  (export "memory" (memory 0))
  (export "__main_argc_argv" (func $__main_argc_argv))
  (export "__data_end" (global 1))
  (export "__heap_base" (global 2))
  (start 4)
  (elem $e0 (i32.const 1) $f5)
  (data $d0 "Exit\0a\00Enter thread\0a\00Failed to create thread\0a\00Failed to join thread %d.\0a\00"))
```

Meanwhile in Release mode, it actually invoke this command:

```
 /opt/wasi-sdk/bin/clang --target=wasm32 --sysroot=/home/hu/wasm-micro-runtime/wamr-sdk/app/libc-builtin-sysroot -nostdlib -z stack-size=4096 -O3 -DNDEBUG -Wl,--initial-memory=65536,--no-entry,--strip-all,--allow-undefined-file=/home/hu/wasm-micro-runtime/wamr-sdk/app/libc-builtin-sysroot/share/defined-symbols.txt    -Xlinker --export=__heap_base -Xlinker --export=__data_end -Xlinker --export=__main_argc_argv -Xlinker --shared-memory -Xlinker --max-memory=196608 -Xlinker --no-check-features CMakeFiles/lokmachau.wasm.dir/main.c.o -o lokmachau.wasm
```

Jackson

huming2207, 2022-03-09:
Hmmmm, interesting...it looks like with `-O2`, `-O3` or `-Oz` the error will appear. Is that a compiler bug...?

lum1n0us, 2022-03-09:
Assume the release/debug mode is about `CMAKE_BUILD_TYPE`. 

Here is my [example code](https://gist.github.com/lum1n0us/ae782f4936520dca0224ffa89a3376f9). It works well on both release mode and debugs mode. Please take a look and let us know the difference between yours.

```
$  cmake  -DWASI_SDK_DIR=/opt/wasi-sdk -DCMAKE_TOOLCHAIN_FILE=../../../wamr-sdk/app/wamr_toolchain.cmake -DCMAKE_SYSROOT=../../../wamr-sdk/app/libc-builtin-sysroot/ -DCMAKE_BUILD_TYPE=Debug ..
$  cmake  -DWASI_SDK_DIR=/opt/wasi-sdk -DCMAKE_TOOLCHAIN_FILE=../../../wamr-sdk/app/wamr_toolchain.cmake -DCMAKE_SYSROOT=../../../wamr-sdk/app/libc-builtin-sysroot/ -DCMAKE_BUILD_TYPE=Release ..
```

huming2207, 2022-05-04:
Hi @xujuntwt95329 @lum1n0us 

Sorry for the late reply. I've been called to do something else at work earlier and I forgot to trace up this issue.

I've upgraded my machine at home to Ubuntu 22.04 and tried again with LLVM/Clang toolchain from Ubuntu official repo, by command `apt install clang llvm lld`, and magically the CMake release mode (with -Oz flag) works. Here's the WASM it generates:

```wasm
(module
  (type $t0 (func (param i32) (result i32)))
  (type $t1 (func (param i32)))
  (type $t2 (func (param i32 i32 i32 i32) (result i32)))
  (type $t3 (func (param i32 i32) (result i32)))
  (type $t4 (func))
  (import "env" "puts" (func $env.puts (type $t0)))
  (import "env" "pthread_exit" (func $env.pthread_exit (type $t1)))
  (import "env" "pthread_create" (func $env.pthread_create (type $t2)))
  (import "env" "pthread_join" (func $env.pthread_join (type $t3)))
  (import "env" "printf" (func $env.printf (type $t3)))
  (func $f5 (type $t4)
    block $B0
      block $B1
        block $B2
          i32.const 1096
          i32.const 0
          i32.const 1
          i32.atomic.rmw.cmpxchg
          br_table $B2 $B1 $B0
        end
        i32.const 1024
        i32.const 0
        i32.const 69
        memory.init $d0
        i32.const 1096
        i32.const 2
        i32.atomic.store
        i32.const 1096
        i32.const -1
        atomic.notify
        drop
        br $B0
      end
      i32.const 1096
      i32.const 1
      i64.const -1
      i32.atomic.wait
      drop
    end
    data.drop $d0)
  (func $f6 (type $t0) (param $p0 i32) (result i32)
    i32.const 1029
    call $env.puts
    drop
    i32.const 0
    call $env.pthread_exit
    i32.const 0)
  (func $__main_argc_argv (type $t3) (param $p0 i32) (param $p1 i32) (result i32)
    (local $l2 i32)
    global.get $g0
    i32.const 16
    i32.sub
    local.tee $l2
    global.set $g0
    block $B0
      local.get $l2
      i32.const 12
      i32.add
      i32.const 0
      i32.const 1
      i32.const 0
      call $env.pthread_create
      i32.eqz
      br_if $B0
      i32.const 1042
      call $env.puts
      drop
    end
    block $B1
      local.get $l2
      i32.load offset=12
      i32.const 0
      call $env.pthread_join
      i32.eqz
      br_if $B1
      local.get $l2
      local.get $l2
      i32.load offset=12
      i32.store
      i32.const 1066
      local.get $l2
      call $env.printf
      drop
    end
    i32.const 1024
    call $env.puts
    drop
    local.get $l2
    i32.const 16
    i32.add
    global.set $g0
    i32.const 0)
  (table $T0 2 2 funcref)
  (memory $memory 1 3 shared)
  (global $g0 (mut i32) (i32.const 5200))
  (global $__data_end i32 (i32.const 1100))
  (global $__heap_base i32 (i32.const 5200))
  (export "memory" (memory 0))
  (export "__main_argc_argv" (func $__main_argc_argv))
  (export "__data_end" (global 1))
  (export "__heap_base" (global 2))
  (start 5)
  (elem $e0 (i32.const 1) $f6)
  (data $d0 "Exit\00Enter thread\00Failed to create thread\00Failed to join thread %d.\0a\00"))

```

I still don't know why it didn't work earlier. But at least the issue has been resolved for now. Again, thanks for your help!

Regards,
Jackson

closed 2022-05-04 by huming2207