wasm-micro-runtime/wasm-micro-runtime #1034: Error when running wasm_runtime_create_exec_env in platform Linux-SGX

dzobbe, 2022-03-08:
Hello All,

I am facing an issue while trying to run the wasm_runtime_create_exec_env function in a SGX enclave. I have noticed from the enclave-sample app that the command for creating exec env is not implemented. Is it supported this function in the enclave? 
Here is the code. 

```C++
 os_set_print_function(enclave_print);
    
    char *buffer;
    std::array<char, 128> error_buf;
    uint32_t size, stack_size = 8092, heap_size = 8092;

    auto wasm_initialized = wasm_runtime_init();

    if (!wasm_initialized) {
        xeprintf("Failed to init wasm! \n" );
        return;
    }

    auto module = wasm_runtime_load(wasm_file_buf, wasm_file_size,
                                          error_buf.data(), error_buf.size());

   
    if (!module) {
        xeprintf("Failed to load bytecode! \n" );
        return;
    }

    auto module_inst = wasm_runtime_instantiate(module, stack_size, heap_size, error_buf.data(), error_buf.size());

    if (!module_inst) {
        xeprintf("Failed to instantiate module! \n" );
        return;
    }

    auto exec_env = wasm_runtime_create_exec_env(module_inst, stack_size);

    if (!exec_env) {
        xeprintf("Failed to create env! \n" );
        return;
    }

    auto func = wasm_runtime_lookup_function(module_inst, "hello", nullptr);

    if (!func) {
        xeprintf("Failed to lookup function! \n" );
        return;
    }

    // Setting up the function argument.  We need to copy data in to the WASM sandbox
    // (on account of it being, you know, a /sandbox/).
    std::string arg("C++");
    auto wasm_buf = wasm_runtime_module_dup_data(module_inst, arg.data(), arg.size());
    std::array<uint32_t, 2> func_args = {wasm_buf, arg.size()};

    auto call_succeeded = wasm_runtime_call_wasm(exec_env, func, func_args.size(), func_args.data());

    if (!call_succeeded) {
        xeprintf("Failed to call function! \n" );
        return;
    }

    // The return value is in the first element of the function argument array.
    // It's a WASM pointer to a null-terminated string.  Fortunately, WAMR allows
    // us to map WASM pointers to native pointers.
    auto func_rv = wasm_runtime_addr_app_to_native(module_inst, func_args[0]);

    xprintf("WASM function returned: %s\n\n",std::string(static_cast<char*>(func_rv)).c_str());
```

wenyongh, 2022-03-09:
Hi, the wasm_runtime_create_exec_env calls wasm_exec_env_create to create the exec_env, normally we should be able to call them in the enclave. Although in enclave-sample app the command for creating exec env hasn't been implemented, but the command CMD_EXEC_APP_FUNC/CMD_EXEC_APP_MAIN will call  wasm_application_execute_main/wasm_application_execute_main, which eventually call wasm_exec_env_create.

Could you please add logs (e.g. os_printf(...)) in wasm_exec_env_create to check the details?

wenyongh, 2022-03-10:
Yes, by default the building of linux-sgx enables the lib-pthread support, which might need to set aux stack while creating exec env. Do you use wasi-sdk to compile the wasm app? Maybe you can try adding `-Wl,--export=__heap_base,--export=__data_end` when building wasm app and run it again wit iwasm of default built (without disable WASM_ENABLE_THREAD_MGR).