wasm-micro-runtime/wasm-micro-runtime #1038: Bug: atomic wait waits microsecond instead of nanoseconds

kfessel, 2022-03-09:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/006574307557b4e041eeb785274021027e338ade/core/iwasm/common/wasm_shared_memory.c#L295

is used to implement [memory.atomic.wait](https://github.com/WebAssembly/threads/blob/master/proposals/threads/Overview.md#wait) the ops timeout should be nanoseconds  but it is interpreted in microseconds

https://github.com/bytecodealliance/wasm-micro-runtime/blob/006574307557b4e041eeb785274021027e338ade/core/iwasm/interpreter/wasm_interp_classic.c#L3197

https://github.com/bytecodealliance/wasm-micro-runtime/blob/006574307557b4e041eeb785274021027e338ade/core/iwasm/common/wasm_shared_memory.c#L377

https://github.com/bytecodealliance/wasm-micro-runtime/blob/006574307557b4e041eeb785274021027e338ade/core/shared/platform/common/posix/posix_thread.c#L236

xujuntwt95329, 2022-03-10:
Hi, thanks for reporting this issue.

I have submitted a PR #1041  to fix this. 

BTW, do you have any case that can be compiled to `atomic.wait` opcode?

kfessel, 2022-03-10:
> BTW, do you have any case that can be compiled to `atomic.wait` opcode?

No, i was just checking, Why i have that semaphore in the riot port, tracked it back and found it not matching the wasm specs.

xujuntwt95329, 2022-03-11:
@kfessel Thanks a lot!

The PR #1041 has been merged

closed 2022-09-27 by wenyongh