wasm-micro-runtime/wasm-micro-runtime #1039: Support more socket APIs

lum1n0us, 2022-03-10:
Plan to support three more socket APIs

- ` int socketpair(int domain, int type, int protocol, int sv[2]);`
- `ssize_t sendmsg(int sockfd, const struct msghdr *msg, int flags);`
- ` ssize_t recvmsg(int sockfd, struct msghdr *msg, int flags);`

The socket APIs feature is an experimental feature for now and it is in the
_dev/socket_api_ branch.

To support some APIs usually means two things:

- allow the compiler to translate APIs from C/C++ to WASM. It needs WASM
  implementations of those APIs. Because it happens before compilation(from
  c/c++ to wasm), the implementation is still in c/c++. [wasi-libc](https://github.com/WebAssembly/wasi-libc)
  is for _libc_ as an example.
- runtime support for APIs. Some functions are so simple that they are able
  to be fully implemented in WASM. Others may require _syscall_ have to depend
  on runtime support. An imported WASM function provided by runtime is a bridge.

So, when going to support a API, we shall ask the question: Can the API be
implemented with existing APIs? If yes, it means we can give the API a fully
WASM implementation. If no, it means we need to add relative capability into
the runtime.

Take `socketpair()` as an example. It is a [wrapper](https://code.woboq.org/userspace/glibc/sysdeps/unix/sysv/linux/socketpair.c.html)
of a `SOCKETCALL`. A *syscall* means additional runtime support. So its 
implementation should include:
- an WASM implementation
- an imported WASM function as bridge
- an additional runtime support

Please take a look at 
- core/iwasm/libraries/lib-socket/
- samples/socket-api/

Let's talk how to handle *marshal* between C/C++ and WASM.

First thing is always *type*. Those WASM implementation have their own types, 
like `__wasi_iovec_t`, and they create and initialize values of `__wasi_XX` types
according to original arguments of c/c++ types. For example, the WASM implementation 
of `sendmsg()` should translate `struct msghdr* msg` to `__wasi_iovec_t s_data`.
Then WASM implementations work with `__wasi_xx` types. 

Assume there is a bridge and additional runtime support is envolved.  Since 
runtime will use platform implementation,  those `__wasi_xx` types need to be
[translate back](core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c) to 
original, required c/c++ types.

Second thing is function signatures. For security, we tends not to use the
exactally *errno* between WASM and native. Instead of it, all WASM implementations,
includes imported ones, always return error number (`__wasi_errno_t`) and use
an output pointer to carry the original return value.

Good luck and don't hesitate to ask when you have any question.

lum1n0us, 2022-03-10:
@BenSunWhiteBoard. Let's do it.

lum1n0us, 2022-03-10:
Here is [a PR](https://github.com/bytecodealliance/wasm-micro-runtime/pull/1042) about the issue that includes an example for testing.

BenSunWhiteBoard, 2022-03-10:
While I am looking at headers and c source code in directory `core/iwasm/libraries/lib-socket/`, I can't help but thinking a unrelated question, Do we have a style guide document that I can read or an auto formatter config file? It's just a little different from what I am used to before, I am kinda wondering what the criteria is.

lum1n0us, 2022-03-11:
Please use `$ clang-format --style=file` under the *wasm-micron-runtime*. It will apply a predefined code style in [*.clang-format*](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/.clang-format).

hangedfish, 2022-03-28:
Is there any plan to support `getaddrinfo` ?

lum1n0us, 2022-03-30:
We might support it in months. A real use case will help to level up the task priority. 😺

closed 2022-11-12 by lum1n0us

loganek, 2022-11-12:
@hangedfish getaddrinfo is already supported, see https://github.com/bytecodealliance/wasm-micro-runtime/issues/1336 for details