wasm-micro-runtime/wasm-micro-runtime #1043: [wasm-c-api] export C function with given name and module name

pengyz, 2022-03-10:
I use AssemblyScript and wamr to build my application.

I found this doc [Export Native API](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md) and followed, now I can register my C function to the wamr runtime like this:

```C
static NativeSymbol g_native_symbols[] = {
    EXPORT_WASM_API_WITH_SIG(add, "(ii)i"),
    EXPORT_WASM_API_WITH_SIG(print_string, "($)"),
};
```
and use it in AssemblyScript as this:
```ts
// @ts-ignore
@external("env", "print_string")
declare function print_string(a: ArrayBuffer): void;

// @ts-ignore
@external("env", "add")
declare function add(a: i32, b: i32): i32;
```

Then I found the **wasm c api** and try do the same thing use it:
```C
own wasm_functype_t* print_string_type = wasm_functype_new_1_0(wasm_valtype_new(WASM_I32));
own wasm_func_t* print_string_func =
    wasm_func_new_with_env(store, print_string_type, print_string, store, NULL);

wasm_functype_delete(print_string_type);

// Instantiate.
printf("Instantiating module...\n");
wasm_extern_vec_t imports;
wasm_extern_t* data[] = {wasm_func_as_extern(print_string_func)};
wasm_extern_vec_new(&imports, 1, data);

own wasm_instance_t* instance = wasm_instance_new(store, module, &imports, NULL);
```
But it won't work, I get the wrong function called.
So how to provide the module name and function name when I export C function using the **wasm c api**?
The wasm_func_t struct has module_name and name field:

```C
struct wasm_func_t {
    wasm_store_t *store;
    wasm_name_t *module_name;
    wasm_name_t *name;
    uint16 kind;

    struct wasm_host_info host_info;
    wasm_functype_t *type;

    bool with_env;
    union {
        wasm_func_callback_t cb;
        struct callback_ext {
            void *env;
            wasm_func_callback_with_env_t cb;
            void (*finalizer)(void *);
        } cb_env;
    } u;
    /*
     * an index in both functions runtime instance lists
     * of interpreter mode and aot mode
     */
    uint16 func_idx_rt;
    WASMModuleInstanceCommon *inst_comm_rt;
    WASMFunctionInstanceCommon *func_comm_rt;
};
```
But wasm_func_new_with_env and wasm_func_new won't set it and they are always null pointer.
I try to set it manually like this:
  ```C
wasm_name_t func_name;
wasm_name_new_from_string(&func_name, "print_string");
wasm_name_t module_name;
wasm_name_new_from_string(&module_name, "env");
print_string_func->name = &func_name;
print_string_func->module_name = &module_name;
```
But it still won't work. They are just ignored in current implementation.

There is my question:
Can I give module name and function name when export a C function using wasm c api ? Is it just unimplement or unsupported by design?

Thanks for reading my question.

pengyz, 2022-03-11:
I found a c api example in wasmer c api: [wasmer-wasi-c-api-example.c](https://github.com/wasmerio/wasmer-c-api/blob/master/wasmer-wasi-c-api-example.c), it support provide module name and function name when exporting c api like this:

```C
// Create a new func to hold the parameter and signature
// of our `host_print` host function
wasmer_value_tag params_sig[] = {WASM_I32, WASM_I32};
wasmer_value_tag returns_sig[] = {};
wasmer_import_func_t *func = wasmer_import_func_new((void (*)(void *)) host_print, params_sig, 2, returns_sig, 0);

// Create module name for our imports
// represented in bytes for UTF-8 compatability
const char *module_name = "env";
wasmer_byte_array module_name_bytes = { .bytes = (const uint8_t *) module_name,
                                        .bytes_len = strlen(module_name) };

// Define a function import
const char *import_name = "host_print";
wasmer_byte_array import_name_bytes = { .bytes = (const uint8_t *) import_name,
                                        .bytes_len = strlen(import_name) };
wasmer_import_t func_import = { .module_name = module_name_bytes,
                                .import_name = import_name_bytes,
                                .tag = WASM_FUNCTION,
                                .value.func = func };

// Define a memory import
const char *import_memory_name = "memory";
wasmer_byte_array import_memory_name_bytes = { .bytes = (const uint8_t *) import_memory_name,
                                               .bytes_len = strlen(import_memory_name) };
wasmer_import_t memory_import = { .module_name = module_name_bytes,
                                  .import_name = import_memory_name_bytes,
                                  .tag = WASM_MEMORY };
wasmer_memory_t *memory = NULL;
wasmer_limit_option_t max = { .has_some = true,
                              .some = 256 };
wasmer_limits_t descriptor = { .min = 256,
                               .max = max };
```

So the wasm c api in wasmer support provide module name and function name as the code shown ?
Prefer our c api provide the same functionality when exporting c function to wasm side.

lum1n0us, 2022-03-11:
`wasm_instance_new()` will combine `imports`(the third argument) with all imported items of a WASM module. The whole process is based on position, not names. For example, `imports[0]` links with the first imported item, `imports[1]` links with the second one.

Usually, the below things will cause failures:

- `imports` doesn't have enough elements to match all required imported items of a WASM module.
- `imports` doesn't have the exact order that a WASM module requires. For example, 
  - the first requirement is a global, but `imports[0]` is a function
  - the second requirement is a function with two arguments, but `imports[1]` is a function with one argument

Hope it is helpful.

pengyz, 2022-03-11:
> `wasm_instance_new()` will combine `imports`(the third argument) with all imported items of a WASM module. The whole process is based on position, not names. For example, `imports[0]` links with the [...]

Very thanks for your anwser ! 
I will follow your advice in my AssemblyScript code to see if it fixed the problem.

lum1n0us, 2022-03-11:
One more thing, please make sure the order of imported items with the .wasm itself instead of the AssemblyScrtpt code. 

*wasm-objdump* of *WABT* will be helpful.

pengyz, 2022-03-11:
> One more thing, please make sure the order of imported items with the .wasm itself instead of the AssemblyScrtpt code.
> 
> _wasm-objdump_ of _WABT_ will be helpful.

You are right, the imported item order in .wat is different from the order in AssemblyScript. 

```ts
// @ts-ignore
@external("env", "print_string")
declare function print_string(a: ArrayBuffer): void;

// @ts-ignore
@external("env", "add")
declare function add(a: i32, b: i32): i32;
```

```wat
 (import "env" "add" (func $assembly/ferryBridge/add (param i32 i32) (result i32)))
 (import "env" "print_string" (func $assembly/ferryBridge/print_string (param i32)))
```

So the order based import is hard to use in AssemblyScript beacuse I can't specify the import function order in the generated wasm.
Consider I have many function exported to wasm, after add a new export function, I'll ensure the new import order and change my C code to fit it. Is there any way to solve this problem expect to generate the c function according to the wat file?

lum1n0us, 2022-03-11:
unfortunately no.

pengyz, 2022-03-11:
> unfortunately no.

Thans for your answer again.
I'll consider to use **wasm_runtime_register_natives** instead. But import function by name is really import for our wasm c api implementation when working with language such as AssemblyScript.
Hope we will have this ability to do this such as wasmer does.

closed 2022-04-20 by pengyz