wasm-micro-runtime/wasm-micro-runtime #1056: `wasm_runtime_load` unexpectedly modifies the input buffer

lei-april, 2022-03-23:
While embedding WAMR into my application, I encountered a weird bug: when I call `wasm_runtime_load` twice with the same inputs, the first call succeeds but the second fails.

After some investigation, I found out that [`const_str_list_insert`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/80887837757c1f9e487ac92e8c5ab60dff24bd89/core/iwasm/interpreter/wasm_loader.c#L371), which is called by `wasm_runtime_load`, casts away `const` of the input buffer and modifies it. This makes the input wasm module ill-formed and cannot be further used.

Seeing the following signature of `wasm_runtime_load`, I think most people would expect it to only read `buf` but never modify it. But clearly WAMR violates this expectation.
```c
wasm_module_t
wasm_runtime_load(const uint8_t *buf, uint32_t size,
                  char *error_buf, uint32_t error_buf_size);
```

Frankly speaking, I think this is poorly designed, and should be fixed.

wenyongh, 2022-03-23:
@lei-april , thanks for reporting the issue! The original design is that the wasm file could be loaded read-only, but during the feature development, we modified the wasm file to reduce the footprint (for example, re-use the string in the wasm file by moving one byte backward and append `\0` at the end) and improve the performance (only for classic interpreter, besides with changing string, some opcodes are changed). Sorry for the misunderstanding, we will modify the API's description soon.

Just wondering whether the read-only loading for wasm file is a hard requirement for you? Normally you can backup the wasm file and load from the backup file next time. If it is a hard requirement, we may consider adding a new API e.g. wasm_runtime_load_ro to load the wasm file with changing it for fast-interpreter mode and AOT mode. Thanks.

lei-april, 2022-03-23:
> Just wondering whether the read-only loading for wasm file is a hard requirement for you? Normally you can backup the wasm file and load from the backup file next time.

No, it's not a hard requirement. I just think the API is misleading. Casting away `const` is rarely a good practice.

BTW, it seems that lifetime of the input buffer to `wasm_runtime_load` needs to persist until `wasm_runtime_unload` is called. That could also become a trap to the unwary.

wenyongh, 2022-03-24:
> No, it's not a hard requirement. I just think the API is misleading. Casting away `const` is rarely a good practice.
>
> BTW, it seems that lifetime of the input buffer to `wasm_runtime_load` needs to [...]

Yes, sorry for the misleading, we submitted patch #1059 to fix it, hope it is clear now.

lei-april, 2022-03-24:
Thanks for the quick responding! I'll close the issue.

closed 2022-03-24 by lei-april