wasm-micro-runtime/wasm-micro-runtime #1058: Is it necessary to catch SIGSEGV when checking the HW bound?

gnh1201, 2022-03-23:
A system that triggers SIGSEGV when trying to check HW bound. I've mentioned this before. #802

Although this problem can be solved by adding the flag(`-DWAMR_DISABLE_HW_BOUND_CHECK=1`), I don't think I know in advance if the target system has this problem.

Do these segmentation faults usually require a warning in advance warning or handling? I'd like some advice.

wenyongh, 2022-03-24:
@gnh1201 The HW bound check is to use the hardware trap mechanism to implement the wasm memory access boundary check for AOT/JIT: runtime firstly registers a signal handler to catch the SIGSEGV signal, and when the wasm app accesses linear memory with address which is out of range (for example, linear memory size is 64KB, the address to read is larger than 64KB), the signal SIGSEGV will be triggered and caught by the signal handler, and in the signal handler, runtime handles the exception and continues to work. The main goal is to improve the performance, if not, runtime (AOT/JIT code) have to check whether the address is out of range manually, some extra instructions will be needed.

So it is normal that SIGSEGV is thrown when HW bound check is enabled, you can see it in GDB, **but it doesn't mean that runtime runs failed, instead, please let wasm app continue to run (in GDB, run `c`)**, normally the "out of bounds memory access" exception will be thrown by runtime, and you can use wasm_runtime_get_exception to get the result. Could you please try again?

If runtime cannot correctly handle the signal, normally runtime will print `unhandled SIGSEGV, si_addr:`. Thanks.

gnh1201, 2022-05-12:
I finally solved this problem. Thanks for your advice. :)

closed 2022-05-12 by gnh1201