wasm-micro-runtime/wasm-micro-runtime #1080: Issues in Running WAMR SGX in Azure using DCAP drivers

dzobbe, 2022-04-11:
Hello everyone,

I am having issues in deploying the WAMR SGX application in Azure using the DCAP driver. I am getting this error:

"_Enclave not authorized to run, .e.g. provisioning enclave hosted in app without access rights to /dev/sgx_provision. You need add the user id to group sgx_prv or run the app as root._"

Any idea on how to address this?

dzobbe, 2022-04-12:
Yeah that was the first thing I tried but it didn't work

dzobbe, 2022-04-12:
I am running as root in the container. And other SGX apps in the same container work just fine.
If I run the container in a desktop SGX machine without the DCAP driver th sgx-wasm application runs fine.

wenyongh, 2022-04-12:
Not known what the root cause is, may be you can get help from below:
https://github.com/intel/linux-sgx