wasm-micro-runtime/wasm-micro-runtime #1092: possible stack overflow while writing outs_area

versaloon, 2022-04-15:
in wasm_interp_call_func_bytecode:call_func_from_interp, outs_area will be accessed to write function parameters to new frame_lp without stack check.

```c
    call_func_from_interp:
    {
        /* Only do the copy when it's called from interpreter.  */
        WASMInterpFrame *outs_area = wasm_exec_env_wasm_stack_top(exec_env);
        POP(cur_func->param_cell_num);
        SYNC_ALL_TO_FRAME();
        word_copy(outs_area->lp, frame_sp, cur_func->param_cell_num);
        prev_frame = frame;
    }
```
There are maybe other similar code to access outs_ares on stack without check.

wenyongh, 2022-04-15:
Hi, the outs area size is checked when allocating the frame of current function, see:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_exec_env.h#L184
Thanks.

versaloon, 2022-04-15:
OK, I see.

closed 2022-04-15 by versaloon