wasm-micro-runtime/wasm-micro-runtime #1113: Support 3rd-party wasi implementation

versaloon, 2022-04-21:
I've implemented a wasi library for an RTOS. And it works OK with the C demo at [wasi](https://github.com/bytecodealliance/wasmtime/blob/main/docs/WASI-tutorial.md#running-common-languages-with-wasi).

1. I use init_args to set the native wasi module, looks good.
```c
#if WASM_ENABLE_LIBC_VSFWASI != 0
    init_args.native_module_name = "wasi_snapshot_preview1";
    init_args.n_native_symbols = get_libc_wasi_export_apis(&init_args.native_symbols);
#endif

    /* initialize runtime environment */
    if (!wasm_runtime_full_init(&init_args)) {
        printf("Init runtime environment failed.\n");
        return -1;
    }
```
2. Implement wasm_application_execute_main_app which is almost the same as wasm_application_execute_main with modification below:
```c
#if WASM_ENABLE_LIBC_WASI != 0 || WASM_ENABLE_LIBC_VSFWASI != 0
    /* In wasi mode, we should call the function named "_start"
       which initializes the wasi envrionment and then calls
       the actual main function. Directly calling main function
       may cause exception thrown. */
    if ((func = wasm_runtime_lookup_wasi_start_function(module_inst))) {
#if WASM_ENABLE_DEBUG_INTERP != 0
        wasm_runtime_start_debug_instance(exec_env);
#endif
        return wasm_runtime_call_wasm(exec_env, func, 0, NULL);
    }
#endif /* end of WASM_ENABLE_LIBC_WASI */
```
Check wasm_runtime_lookup_wasi_start_function if the 3rd-party wasi library is enabled.
3. Copy wasm_runtime_lookup_wasi_start_function in wasm_runtime_common.c  to main.c, because if WASM_ENABLE_LIBC_WASI is not enabled, this function will not included.

Is there any more graceful way to use a 3rd-party wasi library?

wenyongh, 2022-04-24:
Hi, for the question 2 and 3, seems that you can run iwasm with `-f _start` to call wasi entry function: `iwasm -f _start <wasm/aot file>`.

For question 1, we are planning to implement registering native lib for iwasm, e.g., `iwasm --native-lib=<so file>`, developer can build the native APIs into a .so library file, and provide a API like below,
```C
uint32_t
get_native_lib(char **p_module_name, NativeSymbol **p_native_symbols);
```
so that iwasm can call it and get the module_name, native symbol array and native symbol count from the native lib, and then register it.
Not sure whether it is good for your RTOS, if not, normally you should keep your own modification. Thanks.

versaloon, 2022-04-24:
> Hi, for the question 2 and 3, seems that you can run iwasm with `-f _start` to call wasi entry function: `iwasm -f _start <wasm/aot file>`.
>
> For question 1, we are planning to implement registering [...]

-f option is good for me, I will do a test.
For question 1, current solution is good for me too, but yours is better, I will be able to register more than 1 modules.

versaloon, 2022-04-24:
-f _start is OK with one modification:
```c
static void *
app_instance_func(wasm_module_inst_t module_inst, const char *func_name)
{
    wasm_application_execute_func(module_inst, func_name, 0, NULL);
    /* The result of wasm function or exception info was output inside
       wasm_application_execute_func(), here we don't output them again. */
    return NULL;
}
```
For _start wasm_application_execute_func should not pass argc/argv. ofc, it's OK for per platform product_mini main.c.

I close the issue.

closed 2022-04-24 by versaloon