wasm-micro-runtime/wasm-micro-runtime #1115: Data segment's base offset value was modified makes instantiation failed.

YoungWenMing, 2022-04-22:
Hello, I met a bug when building wasm app based on this runtime.

Bad case:
with this wasm module:
```
(module
  (type (;0;) (func (param i32) (result i32)))
  (import "mod" "offset" (global (;0;) i32))
  (func $load (type 0) (param i32) (result i32)
    local.get 0
    i32.load align=1)
  (memory (;0;) 1 1)
  (export "load" (func $load))
  (data (;0;) (global.get 0) "\05\05\05\05"))
```
if I try to instantiate multiple times based on the module showed above, it may failed with `WASM module instantiate failed: unknown global`.
I found that wasm_runtime modified data segment's raw value, which lead to an out-of-range access to a global value.
Below is a reproducing case based on c-api:
[error.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/8536487/error.txt)
Steps to reproduce:

- add this file(change its suffix to .c) and the wat code to samples/wasm-c-api/src
- and then add this sample to samples/wasm-c-api/CMakeLists.txt
- build c-api samples with `make error`

wenyongh, 2022-04-23:
@YoungWenMing Thanks for reporting the issue and submitting the patch to fix it! We have merged the PR, please test again.

YoungWenMing, 2022-04-29:
LGTM~
close this issue now.

closed 2022-04-29 by YoungWenMing

Zhangyx24, 2022-07-19:
Now it failed in AOT and JIT ?

In JIT mode, after build, execute this:
[image]

In AOT mode, after build, execute this:
[image]