wasm-micro-runtime/wasm-micro-runtime #1141: wamrc: Compiling generated LL files

arjunr2, 2022-05-03:
It looks like wamrc currently supports generating LLVM optimized and unoptimzed code. I am looking to run some instrumentation passes on this LL file and generate an AoT executable again.

However, function labels seem to be discarded in the generated LL file (particularly the main or start function), and it cannot be re-compiled using clang. Is there a way to generate Wasm/AoT code from the generated LL file?

Thanks!

wenyongh, 2022-05-06:
Hi, not sure why you cannot apply the extra passes to the LLVM module but need to apply them to LL file?
To compile LL file to an object file, maybe you can use `clang -o <object file> -c <LL file>`.
To generate the AOT file with the instrumented LL file, maybe you can hack the compilation process by yourself:
Change
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/compilation/aot_emit_aot_file.c#L2679-L2689
to code like:
```C
emit LL file and instrument it
use clang to compile LL file to object file
load object file, read it to obj_data->mem_buf, refer to:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/compilation/aot_emit_aot_file.c#L2663
```
Thanks.

arjunr2, 2022-05-06:
I am curious as to the purpose of the output of the `--format=llvmir-unopt` option for `wamrc` if the LLVM cannot be compiled. I was under the assumption I would be able to run `opt` passes and compile the LL file to WASM. But it seems like that is not possible because it changes all the function definitions with an `exec_env` parameter

Thanks

wenyongh, 2022-05-10:
Hi, currently the output LL file is for debug purpose only.