wasm-micro-runtime/wasm-micro-runtime #1144: `wasm_runtime_atomic_wait` is not thread-safe

hangedfish, 2022-05-05:
Host App
```C++
// ...

auto main_exec_env = wasm_runtime_create_exec_env(module_inst, stack_size);
if (main_exec_env == nullptr) {
    std::cout << wasm_runtime_get_exception(module_inst) << std::endl;
    return 3;
}

std::thread t1([&](){
    std::this_thread::sleep_for(std::chrono::seconds(5));
    auto env = wasm_runtime_spawn_exec_env(main_exec_env);
    std::this_thread::sleep_for(std::chrono::seconds(2));
    wasm_runtime_destroy_exec_env(env);
});

std::thread t2([&](){
    std::this_thread::sleep_for(std::chrono::seconds(5));
    auto env = wasm_runtime_spawn_exec_env(main_exec_env);
    std::this_thread::sleep_for(std::chrono::seconds(2));
    wasm_runtime_destroy_exec_env(env);
});

t1.join();
t2.join();
// ...
```
one of  subthreads call `wasm_runtime_spawn_exec_env` return nullptr, and get log
```
[15:32:26:875 - 16FF13000]: HashMap insert elem failed: duplicated key found.
```

Wasm file is enable pthread. 

backtrace in `bh_hashmap.c:95`
```
* thread #3, stop reason = breakpoint 1.1
  * frame #0: 0x000000010000d05c app`bh_hash_map_insert(map=0x0000000100304290, key=0x000000010200616c, value=0x0000600002604000) at bh_hashmap.c:95:13
    frame #1: 0x000000010002aa6c app`acquire_wait_info(address=0x000000010200616c, create=true) at wasm_shared_memory.c:251:14
    frame #2: 0x000000010002a6cc app`wasm_runtime_atomic_wait(module=0x0000000100504080, address=0x000000010200616c, expect=1, timeout=-1, wait64=false) at wasm_shared_memory.c:326:17
    frame #3: 0x0000000100031ba8 app`wasm_interp_call_func_bytecode(module=0x0000000100704080, exec_env=0x0000000120020000, cur_func=0x0000000120008240, prev_frame=0x0000000120020108) at wasm_interp_fast.c:3156:31
    frame #4: 0x000000010003505c app`wasm_interp_call_wasm(module_inst=0x0000000100704080, exec_env=0x0000000120020000, function=0x0000000120008240, argc=0, argv=0x0000000000000000) at wasm_interp_fast.c:3840:9
    frame #5: 0x0000000100048cb8 app`wasm_call_function(exec_env=0x0000000120020000, function=0x0000000120008240, argc=0, argv=0x0000000000000000) at wasm_runtime.c:1727:5
    frame #6: 0x0000000100048e24 app`wasm_create_exec_env_and_call_function(module_inst=0x0000000100704080, func=0x0000000120008240, argc=0, argv=0x0000000000000000, enable_debug=false) at wasm_runtime.c:1763:11
    frame #7: 0x0000000100048760 app`execute_start_function(module_inst=0x0000000100704080) at wasm_runtime.c:952:12
    frame #8: 0x00000001000473b4 app`wasm_instantiate(module=0x00000001004041f0, is_sub_inst=true, stack_size=2097152, heap_size=0, error_buf=0x0000000000000000, error_buf_size=0) at wasm_runtime.c:1550:13
    frame #9: 0x0000000100026258 app`wasm_runtime_instantiate_internal(module=0x00000001004041f0, is_sub_inst=true, stack_size=2097152, heap_size=0, error_buf=0x0000000000000000, error_buf_size=0) at wasm_runtime_common.c:863:44
    frame #10: 0x000000010004cb80 app`wasm_cluster_spawn_exec_env(exec_env=0x0000000110120000) at thread_manager.c:390:29
    frame #11: 0x000000010002946c app`wasm_runtime_spawn_exec_env(exec_env=0x0000000110120000) at wasm_runtime_common.c:4008:12
```

it look like the `wasm_runtime_atomic_wait` is not thread-safe

xujuntwt95329, 2022-05-06:
Hi, thanks for reporting this issue, seems the wasm_runtime_atomic_wait is not protected by a lock, we've fixed this in PR #1146 
Could you please update the code and try again?

Besides, I noticed that the `start` function inside wasm module is invoked during `spawn_exec_env` according to the callstack, I need to check if this is the correct behavior. Thanks a lot for providing the callstack, it is really useful.

hangedfish, 2022-05-07:
> Hi, thanks for reporting this issue, seems the wasm_runtime_atomic_wait is not protected by a lock, we've fixed this in PR #1146 Could you please update the code and try again?
>
> Besides, I noticed [...]

Confirm resolved

closed 2022-05-07 by hangedfish