wasm-micro-runtime/wasm-micro-runtime #1145: What's the currently amount of addressable memory in the linear memory ?

JamesMenetrey, 2022-05-05:
Dear WAMR developers,

I have a question regarding the amount of memory that Wasm applications can allocate in an instance of linear memory. By the specifications, we may allocate up to 2^16 pages of 2^16 bytes in total, resulting in 4 GB of addressable memory.

Can wasm applications indeed allocate _(more or less)_ 4 GB of memory with the current implementation of WAMR in the linear memory? Also, do you support (or plan to) proposals like [memory64](https://github.com/WebAssembly/memory64/) in the near future?

Many thanks!
Cheers

wenyongh, 2022-05-07:
Hi, currently the maximum page count that wasm app can apply is `2^16 -1 = 65535`, or `4GB - 65536bytes`.
For the memory64 support, we may evaluate it in the future, but currently there is no plan to support it yet.
Thanks.

JamesMenetrey, 2022-05-07:
Many thanks for your answer!

closed 2022-05-07 by JamesMenetrey