wasm-micro-runtime/wasm-micro-runtime #1161: Are there more detailed documentation/specification?

jmgurney, 2022-05-09:
This project looks interesting, but at a first glace, it looks very under specified/documented.  Are there more complete documentation that I'm not seeing?

If so, that'd be great.

If not, what is the documentation road map looking like?

Thanks.

lum1n0us, 2022-05-10:
In the current phase, [ReadMe](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/README.md) and documents under [doc](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/doc) are all our reading materials. There are some interesting [samples](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples). Hope they can be helpful.

jmgurney, 2022-05-10:
I was hoping that there was a place where I could find out exactly what units interval is measured in w/o having to read the code:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/app-framework/sensor/app/wa-inc/sensor.h#L60

I did look at doc, and doc told me to look at the headers for the API, BUT as you can see, the headers are not enough to understand/use the API.