wasm-micro-runtime/wasm-micro-runtime #1168: Support importing memory & table

YoungWenMing, 2022-05-12:
Does this project have any plan to implement supports for importing memories and tables when instantiating a WebAssembly Instance?
Besides, any plan to implement the `grow` method of memories and tables?

lum1n0us, 2022-05-16:
Currently, We prefer the below strategies for safety:
 
- not to use host created memory and table 
- not to grow internal memory and table

remko, 2022-05-22:
@lum1n0us Can you clarify your answer?

I also need to import the table and memory from a different module in an instantiation of a different module. Is there a way to achieve this? Will there ever be one? 

Thanks!

lum1n0us, 2022-05-24:
Unfortunately, there is no way to achieve that for now. Every module has to stick with its own *table* and *memory* for now in WAMR.

Zzzode, 2022-06-06:
> * not to grow internal memory and table

What safety concerns are involved with growing Memory and Table?

lum1n0us, 2022-07-28:
WAMR actually doesn't want host functions linking "host-made" tables and memories. Worries of growing are related to [WAMR memory model](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/memory_tune.md). WAMR will help  .wasm modules to save their memory consumption and might have a few differences with the linear memory in users' minds.

remko, 2022-07-28:
@lum1n0us In my case, my memory and table isn’t host-made. They’re simply imported from another module.

lum1n0us, 2022-07-28:
That is the point. As a runtime, it can't tell an imported memory or table(in `imports` of `wasm_instance_new()`) comes from `wasm_memory_new` and `wasm_table_new` or from another module.

yamt, 2022-08-10:
> Currently, We prefer the below strategies for safety:
> 
>     * not to use host created memory and table
> 
>     * not to grow internal memory and table

i'm not sure if i understand the "safety" argument.
it's just a lack of functionality.

peter-jerry-ye, 2024-09-13:
May I ask if this is supported now? If not, how do we claim that WAMR is '- Full compliant to the W3C Wasm MVP' in README?

lum1n0us, 2024-11-15:
https://github.com/bytecodealliance/wasm-micro-runtime/issues/3807