wasm-micro-runtime/wasm-micro-runtime #1194: How to get real fd in Native API

hangedfish, 2022-05-27:
Hi, I am trying wrap OpenSSL API to Native API.

`SSL_ set_ Fd` requires fd as a parameter，but WAMR provides fd in the sandboxes environment, not real fd.

``` C
int SSL_set_fd(SSL *ssl, int fd);
```

I think it is necessary to provide a helper api for converting sandboxes fd to real fd.

wenyongh, 2022-05-31:
Hi, in current WASI design, for security purpose, we had better not expose the real fd and not provide the help api.
If you really need to use it, maybe you can implement it by yourself, for example, in
`core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c`,
add the code like below:
```C
typedef struct WASIContext {
    struct fd_table *curfds;
    struct fd_prestats *prestats;
    struct argv_environ_values *argv_environ;
    struct addr_pool *addr_pool;
    char *argv_buf;
    char **argv_list;
    char *env_buf;
    char **env_list;
} * wasi_ctx_t;

int vfd2fd(wasm_exec_env_t exec_env, wasi_fd_t vfd)
{
    wasm_module_inst_t module_inst = wasm_runtime_get_module_inst(exec_env);
    wasi_ctx_t wasi_ctx = wasm_runtime_get_wasi_ctx(module_inst);
    struct fd_table *curfds;
    struct fd_object *fo;
    __wasi_errno_t err;

    if (!wasi_ctx)
        report error;

    curfds = wasi_ctx->curfds;
    if (!cur_fds)
        report error;

    err = fd_object_get(curfds, &fo, vfd, __WASI_RIGHT_FD_READ | __WASI_RIGHT_FD_WRITE, 0);
    if (err != 0)
        report error;

    fd = fd_number(fo);

    fd_object_release(fo);
    return fd;
}
```
Refer to the following links:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c#L1238-L1245
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c#L49-L69

hangedfish, 2022-06-01:
Thanks a lot.

closed 2022-06-01 by hangedfish