wasm-micro-runtime/wasm-micro-runtime #1209: "header files" missed while building and installing wamr in "product-mini/platforms/{x}"

yaozhongxiao, 2022-06-07:
When I build and install wamr runtime in ${WAMR_ROOT}/product-mini/platforms/{x}
the "include" directory had been missed; 
IMHO, is it better to place the "install()" in runtime_lib.cmake

wenyongh, 2022-06-09:
Hi, the file CMakeLists.txt under product-mini/platforms/{x} is to build the files for that platform only. Could you please try using the file CMakeLists.txt under ${WAMR_ROOT}?

yaozhongxiao, 2022-06-09:
> Hi, the file CMakeLists.txt under product-mini/platforms/{x} is to build the files for that platform onl

thanks for your reply.

IMHO,  Even if the file CMakeLists.txt under product-mini/platforms/{x} is to build the files for that platform only， but it supports the install command,  does it need to guarantee either full features or nothing?
I can install the mini product for libs(.so, .a), but the headers are missed which leads to the installed artifacts is not integrated to be used.

All the above is just my concern, thanks all the same.