wasm-micro-runtime/wasm-micro-runtime #1221: What are the interpreter Mode, JIt and AOT-related codes?

Zhangyx24, 2022-06-10:
I'm a little confused about which dir includes the Interpreter Mode, JIt and AOT related code？

xujuntwt95329, 2022-06-13:
Hi @Zhangyx24

Interpreter mode is under:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/iwasm/interpreter

AoT mode is under:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/iwasm/aot

JIT mode (and AoT compilation) is under:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/iwasm/compilation

besides, the common layer provides unified APIs to the embedder:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/iwasm/common

abc767234318, 2022-06-13:
> Hi @Zhangyx24
>
> Interpreter mode is under: https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/iwasm/interpreter
>
> AoT mode is under: [...]

I have a question, how does WAMR implement JIT?

xujuntwt95329, 2022-06-13:
@abc767234318 
By default, WAMR use LLVM's ORC JIT API to implement the JIT compilation, which is known as `lazy mode`, in this mode, the functions will be compiled when they are first invoked.

We also provide an option `WAMR_BUILD_LAZY_JIT=1/0`, when this is set to 0, it will be `eager mode`, in this mode, all the functions will be compiled during startup.

abc767234318, 2022-06-13:
@xujuntwt95329 Thanks for your answer!

closed 2022-06-13 by Zhangyx24

hungryzzz, 2023-02-23:
> @abc767234318 By default, WAMR use LLVM's ORC JIT API to implement the JIT compilation, which is known as `lazy mode`, in this mode, the functions will be compiled when they are first invoked.
>
> We [...]

Hi, once the `.wasm` file is compiled to the target machine code, will this target code be optimized during execution? (Just like the multi-tier JIT in JSC)