wasm-micro-runtime/wasm-micro-runtime #1222: Issue about API open  when using WASI

donghengqaz, 2022-06-10:
I am testing the WASI by file-system example, and find an issue about API `open`, my steps are following:

1. build `iwasm`
2. build file-system example by enable stdlib(without `-nostdlib`)
3. run `iwasm` to load file-system example

The result is `errno=76`, but I compile it by gcc and run it well. The file-system example code is following:

```c
#include <stdio.h>
#include <fcntl.h>
#include <unistd.h>
#include <errno.h>

#define BUFFER_SIZE 32

void on_init(void)
{
    int fd;
    int ret;
    const char file[] = "hello.txt";
    const char text[] = "Hello World!";
    int text_size = sizeof(text);
    char buffer[BUFFER_SIZE] = { 0 };

    printf("Opening file %s for writing", file);
    fd = open(file, O_WRONLY | O_CREAT | O_TRUNC);
    if (fd < 0) {
        printf(" failed, errno=%d.\n", errno);
        return;
    } else {
        printf(" OK, fd=%d.\n", fd);
    }

    printf("Write text \"%s\" total %d bytes into file", text, text_size);
    ret = write(fd, text, text_size);
    if (ret < 0) {
        printf(" failed errno=%d.\n", errno);
        close(fd);
        return;
    } else {
        printf(" OK.\n");
    }

    printf("Close file");
    ret = close(fd);
    if (ret < 0) {
        printf(" failed errno=%d.\n", errno);
        return; 
    } else {
        printf(" OK.\n");
    }

    printf("Opening file %s for reading", file);
    fd = open(file, O_RDONLY);
    if (ret < 0) {
        printf(" failed errno=%d.\n", errno);
        return; 
    } else {
        printf(" OK, fd=%d.\n", fd);
    }

    printf("Read text from file");
    ret = read(fd, buffer, BUFFER_SIZE - 1);
    if (ret <= 0) {
        printf(" failed errno=%d.\n", errno);
        close(fd);
        return;
    } else {
        printf(" OK, read text is \"%s\".\n", buffer);
    }

    printf("Close file");
    ret = close(fd);
    if (ret < 0) {
        printf(" failed errno=%d.\n", errno);
        return; 
    } else {
        printf(" OK.\n");
    }

    printf("Opening file %s for seeking", file);
    fd = open(file, O_RDONLY);
    if (ret < 0) {
        printf(" failed errno=%d.\n", errno);
        return; 
    } else {
        printf(" OK, fd=%d.\n", fd);
    }

    printf("Seek file end");
    ret = lseek(fd, 0, SEEK_END);
    if (ret < 0) {
        printf(" failed errno=%d.\n", errno);
        close(fd);
        return;
    } else {
        printf(" OK, file size is %d.\n", ret);
    }

    printf("Close file");
    ret = close(fd);
    if (ret < 0) {
        printf(" failed errno=%d.\n", errno);
        return; 
    } else {
        printf(" OK.\n");
    }
}

int main(int argc, char *argv[])
{
    on_init();

    return 0;
}

```

NingW101, 2022-06-10:
Hey @donghengqaz, you should run `iwasm` with `--dir` option when invoking filesystem api.
```c

iwasm --dir=. test.wasm

```
Could you please have a try?

And here is a document for describing the [`wasi-capabilities`](https://github.com/bytecodealliance/wasmtime/blob/main/docs/WASI-capabilities.md) which you can refer to.

donghengqaz, 2022-06-13:
> Hey @donghengqaz, you should run `iwasm` with `--dir` option when invoking filesystem api.
>
> ```c
> iwasm --dir=. test.wasm
> ```
>
> Could you please have a try?
>
> And here is a document for describing the [...]

Thanks @NingW101 , I will try this.

closed 2022-09-22 by wenyongh