wasm-micro-runtime/wasm-micro-runtime #1230: [Fast Interp] A crash caused by memory corruption in fast interp

jhe33, 2022-06-15:
hi

last week, @lei-april reported a crash when enabled fast interp, the case is [test_stddef_cpp_5_1.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/8908566/test_stddef_cpp_5_1.wasm.zip)

I did an simple investigation and thought the following code doesn't take the correct value to reset the variable. it should be dynamic_offset or start_dynamic_offset instead. Maybe I'm wrong so that please take a double confirm. thank you.
p.s @lei-april said the change passed the spec test already.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/e0a8aa09be9c4af4e3c9c3db6148901a18a675fa/core/iwasm/interpreter/wasm_loader.c#L5041

xujuntwt95329, 2022-06-16:
Hi @jhe33 

Thanks for reporting this issue, line 5041 is not the root cause, the issue is from the DROP opcode, we've submitted a patch to fix this: https://github.com/bytecodealliance/wasm-micro-runtime/pull/1231

Could you please update the code and try again?

jhe33, 2022-06-16:
got it, thank u.

lei-april, 2022-06-16:
> Thanks for reporting this issue, line 5041 is not the root cause, the issue is from the DROP opcode, we've submitted a patch to fix this: #1231

I looked at the patch, but found the following change in `wasm_mini_loader.c` a bit suspicious:
```diff
-                        if (*(loader_ctx->frame_offset)
-                            > loader_ctx->start_dynamic_offset)
-                            loader_ctx->dynamic_offset -= 2;
+                        if ((*(loader_ctx->frame_offset)
+                             > loader_ctx->start_dynamic_offset)
+                            && (*(loader_ctx->frame_offset)
+                                < loader_ctx->max_dynamic_offset))
+                            loader_ctx->dynamic_offset--;
```

Are you sure there's not a typo? @xujuntwt95329

xujuntwt95329, 2022-06-16:
@lei-april Thanks a lot!
Yes, it is a typo, I've submitted a new PR to fix it.

wenyongh, 2022-09-23:
Close as issue was fixed.

closed 2022-09-23 by wenyongh