wasm-micro-runtime/wasm-micro-runtime #1262: Hardware bound check suppresses previous signal handlers

lei-april, 2022-07-01:
When WAMR registers signal handlers for `SIGSEGV` & `SIGBUS`, it doesn't preserve previous handlers for those signals (the third argument of `sigaction`):
```
    if (sigaction(SIGSEGV, &sig_act, NULL) != 0
        || sigaction(SIGBUS, &sig_act, NULL) != 0) {
        os_printf("Failed to register signal handler\n");
        goto fail3;
    }
```

Some applications might use signal handling for other purposes, like reporting stack traces when crashing. Embedding WAMR
 into such an application would make its original signal handling mechanism stop working, unless hardware bound check is disabled.

I looked at wasmtime, and it seems they are paying respect to previous handlers: https://github.com/bytecodealliance/wasmtime/blob/main/crates/runtime/src/traphandlers/unix.rs#L67

It would be great if WAMR could do the same :)

wenyongh, 2022-07-05:
Thanks @lei-april, we will try to enable it.

wenyongh, 2022-07-06:
@lei-april I uploaded PR #1268 to fix it, could you please help check whether it is OK?  Thanks.

closed 2022-07-08 by lei-april