wasm-micro-runtime/wasm-micro-runtime #1272: Wrong result of sum.wasm in AOT mode?

Zhangyx24, 2022-07-07:
I tried to execute sum.wasm by C-api, however the interpreter mode get the right result and the AOT mode get the wrong result.
In interpreter mode, excute the following commands:
`mkdir build`
`cd build`
`cmake ..`
`make`
`./sum32`

In AOT mode, excute the following commands:
`mkdir AOTbuild`
`cd AOTbuild`
`cmake -DWAMR_BUILD_INTERP=0 -DWAMR_BUILD_AOT=1 ..`
`make`
`./sum32`

[image]

The C code is :
#include <stdio.h>

#include "wasm_c_api.h"

#define own

int main(int argc, const char* argv[]) {
  // Initialize.
  printf("Initializing...\n");
  wasm_engine_t* engine = wasm_engine_new();
  wasm_store_t* store = wasm_store_new(engine);

  // Load binary.
  printf("Loading binary...\n");
#if WASM_ENABLE_AOT != 0 && WASM_ENABLE_INTERP == 0
  FILE* file = fopen("empty_imports.aot", "rb");
#else
  FILE* file = fopen("sum32.wasm", "rb");
#endif
  if (!file) {
    printf("> Error loading module!\n");
    return 1;
  }

  int ret = fseek(file, 0L, SEEK_END);
  if (ret == -1) {
    printf("> Error loading module!\n");
    fclose(file);
    return 1;
  }

  long file_size = ftell(file);
  if (file_size == -1) {
    printf("> Error loading module!\n");
    fclose(file);
    return 1;
  }

  ret = fseek(file, 0L, SEEK_SET);
  if (ret == -1) {
    printf("> Error loading module!\n");
    fclose(file);
    return 1;
  }

  wasm_byte_vec_t binary;
  wasm_byte_vec_new_uninitialized(&binary, file_size);
  if (fread(binary.data, file_size, 1, file) != 1) {
    printf("> Error loading module!\n");
    fclose(file);
    return 1;
  }
  fclose(file);

  // Compile.
  printf("Compiling module...\n");
  own wasm_module_t* module = wasm_module_new(store, &binary);
  if (!module) {
    printf("> Error compiling module!\n");
    return 1;
  }

  wasm_byte_vec_delete(&binary);

  // Instantiate with non-null but empty imports array.
  printf("Instantiating module...\n");
  wasm_extern_vec_t imports = WASM_EMPTY_VEC;
  own wasm_instance_t* instance =
    wasm_instance_new(store, module, &imports, NULL);
  if (!instance) {
    printf("> Error instantiating module!\n");
    return 1;
  }

  // Run an exported function to verify that the instance was created correctly.
  printf("Extracting export...\n");
  own wasm_extern_vec_t exports;
  wasm_instance_exports(instance, &exports);
  if (exports.size == 0) {
    printf("> Error accessing exports!\n");
    return 1;
  }else{
      printf("Export size : %zu\n", exports.size);
  }

  const wasm_func_t* func1 = wasm_extern_as_func(exports.data[0]);
  if (func1 == NULL) {
    printf("> Error accessing export!\n");
    return 1;
  }

  wasm_module_delete(module);
  wasm_instance_delete(instance);

  printf("Calling func1...\n");
  wasm_val_t args = WASM_EMPTY_VEC;

  wasm_val_t results[1] = { WASM_INIT_VAL };
  wasm_val_vec_t results_vec = WASM_ARRAY_VEC(results);
  wasm_func_call(func1, &args, &results_vec);
  printf("%d\n",results_vec.data[0].of.i32);

  wasm_extern_vec_delete(&exports);

  // Shut down.
  printf("Shutting down...\n");
  wasm_store_delete(store);
  wasm_engine_delete(engine);

  // All done.
  printf("Done.\n");
  return 0;
}

Zhangyx24, 2022-07-07:
The result of AOT mode is 0 ,and expected to be 3.

Zhangyx24, 2022-07-07:
@xujuntwt95329 Thanks for your answer! It seems that I made a stupid mistake.
I want to ask another question that:
WAMR c-api can get and print the i64 type and i32 type result from wasm program, but how to get v128 result and print?
[image]

xujuntwt95329, 2022-07-07:
Unfortunately, it is not supported to print the v128 value through wasm-c-api

Zhangyx24, 2022-07-07:
Thanks a lot !

closed 2022-07-07 by Zhangyx24