wasm-micro-runtime/wasm-micro-runtime #1275: Recursively call `wasm_application_execute_func` under hardware bound check

lei-april, 2022-07-08:
In our use case, a wasm instance might call another instance with the help of host functions, which means `wasm_application_execute_func` is invoked recursively.

When hardware bound check is enabled, the chain of wasm instances in a recursive call all share the same `aot_exec_env` but each has different `module_inst`s. However, currently all instances in a recursive call would share the first instance's `module_inst`, which is apparently wrong.

A possible fix looks like the following:
```diff
diff --git a/core/iwasm/aot/aot_runtime.c b/core/iwasm/aot/aot_runtime.c
index 3a60811..1d37ea8 100644
--- a/core/iwasm/aot/aot_runtime.c
+++ b/core/iwasm/aot/aot_runtime.c
@@ -1609,8 +1609,15 @@ aot_create_exec_env_and_call_function(AOTModuleInstance *module_inst,
         }
     }
 
+    // switch to current `module_inst`
+    WASMModuleInstanceCommon *prev_module_inst = exec_env->module_inst;
+    exec_env->module_inst = (WASMModuleInstanceCommon *)module_inst;
+
     ret = aot_call_function(exec_env, func, argc, argv);
 
+    // restore previous `module_inst`
+    exec_env->module_inst = prev_module_inst;
+
     /* don't destroy the exec_env if it's searched from the cluster */
     if (!existing_exec_env)
         wasm_exec_env_destroy(exec_env);

```

wenyongh, 2022-07-11:
@lei-april it is simple to apply the patch in the runtime, like PR #1283, but it really makes
confusing that runtime should backup the module instance. We are wondering why your
application cannot backup the module instance before calling runtime API, and restore it
after calling? Thanks.

lei-april, 2022-07-11:
> We are wondering why your
application cannot backup the module instance before calling runtime API, and restore it
after calling?

I think we need something like `wasm_runtime_set_module_inst` to achieve this. But there's only `wasm_runtime_get_module_inst` at the moment.

Another point is that this behavior is inconsistent depending on whether hardware bound check is enabled, which might be confusing to users.

wenyongh, 2022-07-11:
@lei-april I discussed with @xujuntwt95329, we prefer to not change the runtime behavior,
but add the API wasm_runtime_set_module_inst as you mentioned.
I uploaded the PR #1286, could you please check whether it meets your requirement?

lei-april, 2022-07-12:
@wenyongh Thanks. I think adding `wasm_runtime_set_module_inst` could solve our problem.

BTW, if I manually create an `exec_env` via `wasm_runtime_create_exec_env`, is it OK to reuse this `exec_env` with different `module_inst`s (with the help of `wasm_runtime_set_module_inst`)?

wenyongh, 2022-07-12:
Normally it should work, one issue that I can find is the auxiliary stack overflow/underflow check, you can refer to link below about the aux stack:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/memory_tune.md

And here is a PR the resolve the similar issue when calling sub module with the same exec_env of main module in multi-module feature:
[ https://github.com/bytecodealliance/wasm-micro-runtime/pull/1279](https://github.com/bytecodealliance/wasm-micro-runtime/pull/1279)

The aux stack info can be recognized by runtime when `-Wl,--export=__heap_base -Wl,--export=__data_end` is added for wasi-sdk/emsdk clang. If it isn't recognized, using the same exec_env for different module instances should be OK. If yes, had better do the backup/replace/store of the aux_stack_bound/bottom for exec_env like the PR above. But for this, it is not so friendly to add more runtime APIs, could you please just include "wasm_runtime_common.h" in your application, and directly access the fields of exec_env?

The other issue that we had better not do this in multi threading, all exec_envs created by the same wasm app are managed by thread manager library, I am not sure whether current exec_env will be accessed by other thread or not.

lei-april, 2022-07-12:
> Normally it should work, one issue that I can find is the auxiliary stack overflow/underflow check, you can refer to link below about the aux stack:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/memory_tune.md

Got it. For wasm modules that don't export globals like `__heap_base`, the auxiliary stack check should have no effect. Am I correct?

closed 2022-07-18 by lei-april