wasm-micro-runtime/wasm-micro-runtime #1280: How to specify the wasm function to execute?

Zhangyx24, 2022-07-09:
[image]
I tried to specify a function on mac `iwasm test.wasm -f=func1` but failed.
What is the corrrect usage?

xujuntwt95329, 2022-07-09:
Hi @Zhangyx24 

As shown in the usage, the `options` should passed in front of the wasm file, otherwise it will be treated as the `args` to your wasm application.

For your case:

``` C
iwasm -f func1 test.wasm
```

Zhangyx24, 2022-07-09:
Thanks a lot, and how to pass parameters for a function, just put the parameters follow the `iwasm -f func1 test.wasm` in the command ?

xujuntwt95329, 2022-07-09:
Yes, e.g. `iwasm -f func1 test.wasm 1 2`

The runtime will check the number of arguments

Zhangyx24, 2022-07-09:
Thanks a lot !

closed 2022-07-09 by Zhangyx24