wasm-micro-runtime/wasm-micro-runtime #1282: Executing wasm in AOT and JIT mode but got wrong answer?

Zhangyx24, 2022-07-09:
### Steps to reproduce
I tried to execute the wasm 
[roltest.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9076346/roltest.txt)
in AOT and JIT mode in macOS and linux but got the answer.

AOT:
`wamrc -o roltest.aot roltest.wasm`
`iwasm -f _main roltest.aot`

JIT:
`cd product-mini/platforms/darwin/JITbuild/iwasm -f _main roltest.wasm` in mac
and `product-mini/platforms/linux/JITbuild/iwasm -f _main roltest.wasm` in linux

### Expected output
4

### Actual output
The actual output is a random number, everytime I excute it I got a different output.

wenyongh, 2022-07-11:
@Zhangyx24 Thanks for reporting the issue! I submitted PR #1285 to fix it, please try again.

Zhangyx24, 2022-07-11:
[image]
 It seems that the commit didn't pass all the tests.
Should I update directly to this COMMIT version locally and test it ?

Zhangyx24, 2022-07-11:
Besides the roltest.txt, the
 [new_rol_test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9081931/new_rol_test.txt) encounters the same problem.
The expected output is 235, however both AOT and JIT got a random number.

wenyongh, 2022-07-11:
> @wenyongh
> 
> [image] It seems that the commit didn't pass all the tests. Should I update directly to this COMMIT version locally and test it ?

It is due to code guideline issue and has been fixed.

> Besides the roltest.txt, the [new_rol_test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9081931/new_rol_test.txt) encounters the same problem. The expected output is 235, however both AOT and JIT got a random number.

Seems it is the same issue, please rebuild wamrc and test again.

Zhangyx24, 2022-07-11:
OK, I'm going to try it again, thanks for your answer !

Zhangyx24, 2022-07-11:
@wenyongh 
I'm sorry that the problem seems not fixed, I still got a random output when I executing roltest.wat in AOT mode.

wenyongh, 2022-07-11:
Have you rebuilt the wamrc and generated the aot file again?

Zhangyx24, 2022-07-11:
Yes.

wenyongh, 2022-07-11:
@Zhangyx24 I tested again in Linux and MacOS, both of your cases work. Could you please check whether this line is executed:
https://github.com/wenyongh/wasm-micro-runtime/blob/fix_aot_rotr_0/core/iwasm/compilation/aot_emit_numberic.c#L695

Or, could you please dump the llvm ir file with command:
```bash
wamrc --format=llvmir-unopt -o test.ll <wasm file>
```

And see whether the content in the end of file is like:
```
func_end:                                         ; preds = %func_begin
  %phi = phi i64 [ 4, %func_begin ]
  ret i64 %phi
}
```

wenyongh, 2022-07-12:
@Zhangyx24 I uploaded another patch set, could you please try again?

Zhangyx24, 2022-07-12:
@wenyongh Thanks a lot ! I'm going to try it again.

wenyongh, 2022-07-12:
Have you rebuilt wamrc and re-generated aot file again? The handling for JIT and AOT should be the same.

Zhangyx24, 2022-07-12:
Yes, I clone the main branch and rebuild wamrc int the following steps:
[image]

Zhangyx24, 2022-07-12:
@wenyongh Sorry, I tried it again. Now it succeed. Thanks a lot!! I'll close this issue.

closed 2022-07-12 by Zhangyx24