wasm-micro-runtime/wasm-micro-runtime #1287: How do I export a native c++ function using iwasm?

jakob999999, 2022-07-11:
Hello, 

I was wondering if someone has successfully exported a c++ native function and could show me an example?

Exporting c functions works fine for me, but c++ functions sometimes do not.

Thanks in advance.

Best,

Jakob

wenyongh, 2022-07-12:
Hi, normally you can export C function, and call C++ function inside it, for example:
```C
void foo_c_plus_plus()
{
    ...
}

extern "C" {

void foo_c(wasm_exec_env_t exec_env)
{
   foo_c_plus_plus();
}

}
```

jakob999999, 2022-07-12:
Hey Wenyong, 
Thanks for the reply. Building a nativeFunction.cpp file into the iwasm file using cmake does not work for me so I was hoping someone was able to get that working instead of calling a c++ function from c code.

jakob999999, 2022-07-12:
@xujuntwt95329 @gnh1201 Yes, that is exactly what I am trying to do but working with libraries would work too. I am stuck on linking the functions using clang++ and wasi-sysroot. Therefore, if someone has done that before and would be able to link me an example where he was able to run any native C++ function, that would save me a lot of time. I would also try to run a C++ function inside C code as a workaround.

jakob999999, 2022-08-04:
So I used extern "C" to wrap the functions and that solved the issue for me.

closed 2022-08-04 by jakob999999