wasm-micro-runtime/wasm-micro-runtime #1289: Fail to set the max memory with c-api ?

Zhangyx24, 2022-07-12:
## Steps to reproduce
I tried to execute the test.wat:
[test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9091706/test.txt) both in command line and c-api. 

(1)command line
`iwasm test.wat`

(2)c-api
The c file is 
[test-in-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9091728/test-in-c.txt)

## Expected result
Allow the memory of 65536, and don't print error.

## Actual result
(1)WAMR command line succeed and didn't print error.
(2)c-api in interpreter mode failed and print:
[image]
c-api in AOT mode succeed.

lum1n0us, 2022-07-18:
nice catching! will fix it soon.

Zhangyx24, 2022-07-18:
Thanks a lot !  Waiting for your fix.

lum1n0us, 2022-07-19:
After a long discussion, we are going to *keep different* results of interpreter and AOT when setting the memory minimum page count with 65536(4GB) in current phase.

The mainly reason on it is that interpreter and AOT are using different design to manage the linear memory boundary. And we are planing to unite both. And when it's done, both mode shall have same responses to 65536.

So, there will be no fix for the failure of meeting `(memory 65536)` in interpreter.

Zhangyx24, 2022-07-19:
Do you mean that after unite both, interpreter and AOT will response same to 65536?
Will they both allow 65536 or don't?

lum1n0us, 2022-07-19:
After uniting both, both interpreter and AOT will allow 65536 pages in the future. But not in the subsequent few releases.

lum1n0us, 2022-07-19:
https://github.com/bytecodealliance/wasm-micro-runtime/issues/1301

Zhangyx24, 2022-07-20:
Thanks for your answer!

closed 2022-07-20 by Zhangyx24