wasm-micro-runtime/wasm-micro-runtime #1290: Calling host functions in c-api error ?

Zhangyx24, 2022-07-13:
## Steps to reproduce
I tried to execute **test.wat**  
[test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9100940/test.txt)
through c-api in **test-in-api.c** 
[test-in-api.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9100941/test-in-api.txt)
.

In test.wat, it **import 2 function from host (c file)** : `assert_eq_i32` and `print_input`. (**This 2 function has 2 different types !**)
#### The right process to import these 2 functions is to import `assert_eq_i32` firstly and import `print_input` secondly. (line 139 in c file)
#### But I import `print_input` firstly and import `assert_eq_i32` secondly. 

## Expected Result
The c api shows error or exception to **alert** that the first function I imported in c file are different from the first import function in wat file in type.

## Actual Result
It alert nothing and regard `print_input` as `assert_eq_i32` and regard `assert_eq_i32` as `print_input` although they has **different function types** and output unexpected output of wat file.

Zhangyx24, 2022-07-18:
Thanks a lot ! Waiting for your fixing.

lum1n0us, 2022-07-25:
we've fixed the issue in the latest commits, dd62b32b201db8e5939e1084757c712145e93264.

wenyongh, 2022-09-22:
Close this issue as it was fixed.

closed 2022-09-22 by wenyongh