wasm-micro-runtime/wasm-micro-runtime #1293: WAMR fail to execute the wasm file with TiDB ?

Zhangyx24, 2022-07-14:
## Test Case
[test.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9110081/test.zip)

This is a wasm file first mentioned in wasmer issues NO.2777.
The program test.wasm which is a Go database (TiDB) is able to run on wasmer.

## Steps to Reproduce
`iwasm run test.wasm`

## Expected Results
Successfully execute the test.wasm and print :
TiDB [pingcap]>

## Actual Results
It was blocked and print nothing.

## Versions and Environment
WAMR 05-18-2022

Operating system: macOS and ubuntu20.04

Architecture: x86_64

wenyongh, 2022-07-15:
Please increase the operand stack size by adding `--stack-size=n` for iwasm, the case can successfully run with
```bash
iwasm --stack-size=512000 --dir=. test.wasm
```

Zhangyx24, 2022-07-15:
Thanks a lot, now it works !

closed 2022-07-15 by Zhangyx24