wasm-micro-runtime/wasm-micro-runtime #1294: Fail to execute through JIT mode ?

Zhangyx24, 2022-07-14:
## Steps to reproduce 
Executing the wat file though AOT and JIT mode ,
[test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9110878/test.txt)
.
#### AOT
`wamrc -o test.aot test.wasm` and then `iwasm test.wasm`

#### JIT
`product-mini/platforms/darwin/JITbuild/iwasm test.wasm`

## Expected result
Successfully load the wat file.

## Actual result
Succeed only in AOT mode, and JIT mode, it outputs :
`WASM module load failed: ref value type requires reference types feature`

wenyongh, 2022-07-15:
Hi, the reference types feature isn't enabled by default, you can build JIT with:
```bash
cd <wamr_root>/product-mini/platforms/linux
mkdir build && cd build
cmake .. -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_REF_TYPES=1
make -j
```
Thanks.

Zhangyx24, 2022-07-15:
Now it works, thanks for your answer !

Zhangyx24, 2022-07-15:
Yes, I have alreday builf

closed 2022-07-15 by Zhangyx24