wasm-micro-runtime/wasm-micro-runtime #1295: Fail to get the `_start` function ?

Zhangyx24, 2022-07-14:
## Steps to reproduce
[test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9111001/test.txt)

There is a `_start` function in the wat file, but if execute the wat file it didn't find the `_start` function by itself.

`wat2wasm test.wat --enable-alll`
`wamrc -o test.aot test.wasm`
`iwasm test.aot`

## Expected output
0

## Actual output
`[21:50:51:552 - 11148EDC0]: Lookup wasi _start function failed: invalid function type.

Exception: lookup the entry point symbol (like _start, main, _main, __main_argc_argv) failed`

But if use `iwasm -f _start test.aot`, it coulld print the correct result 0.

wenyongh, 2022-07-15:
Hi, currently only `_start` function with signature "()" (no arguments and not results) is supported, the `_start` function with return type `v128` is not allowed.

Zhangyx24, 2022-07-15:
Thanks a lot !

closed 2022-07-15 by Zhangyx24