wasm-micro-runtime/wasm-micro-runtime #1297: WASM module load failed: unsupported opcode fc 0a ？

abc767234318, 2022-07-16:
```cpp
#include <assert.h>
#include <stdio.h>

int main(void) {
	char oldname[] = "/scratch/oldfile";
	char newname[] = "/scratch/newfile";

	FILE *oldfile = fopen(oldname, "w");
	assert(oldfile != NULL);
	assert(fclose(oldfile) == 0);

	assert(rename(oldname, newname) == 0);

	FILE *newfile = fopen(newname, "r");
	assert(newfile != NULL);
	assert(fclose(newfile) == 0);

	return 0;
}
```
This is a source code first mentioned in wasmer issues NO.1503.

### Steps to Reproduce
I compiled it using clang provided with the wasi-sdk.
```
clang rename.c -o rename.wasm
```
Then I execute it with iwasm.
```
iwasm --dir=scratch\ .\rename.wasm
```
And I got the following error.
```
WASM module load failed: unsupported opcode fc 0a
```

abc767234318, 2022-07-16:
I used wasm-objdump to find that the opcode of memory.copy is FC 0A.

xujuntwt95329, 2022-07-18:
Hi @abc767234318 
`memory.copy` is from `bulk memory` feature, could you please rebuild iwasm with ` -DWAMR_BUILD_BULK_MEMORY=1` and try again?

abc767234318, 2022-07-18:
Thanks a lot.

closed 2022-07-18 by abc767234318