wasm-micro-runtime/wasm-micro-runtime #1298: Function not implemented in Windows?

abc767234318, 2022-07-16:
```
fn main() -> std::io::Result<()> {
	for entry in std::fs::read_dir("/")? {
		println!("{}", entry?.path().display());
	}
	Ok(())
}
```
This is a source code first mentioned in wasmer issues NO.2618.

### Steps to Reproduce
```
rustup target add wasm32-wasi
cargo build --target wasm32-wasi
iwasm --dir=. demo.wasm
```
And I got the following error.
```
Error: Os { code: 52, kind: Unsupported, message: "Function not implemented" }
```
My test platform is Windows, but when I test on Ubuntu, I get no errors.

abc767234318, 2022-07-19:
[demo.zip](https://githu@wenyongh .com/bytecodealliance/wasm-micro-runtime/files/9141131/demo.zip)
@wenyongh

wenyongh, 2022-07-20:
@abc767234318 The issue is because that in Windows, we use uvwasi library to support the libc-wasi feature, and the uvwasi doesn't support `wasi_fd_readir` in Windows and Android:
https://github.com/nodejs/uvwasi/blob/main/src/uvwasi.c#L1417-L1418
https://github.com/nodejs/uvwasi/blob/main/src/uvwasi.c#L16-L18

Other runtime should also run failed in Windows if it uses uvwasi library. It is a little complex to fix `wasi_fd_read` by ourselves, we had better wait some time to see whether uvwasi will fix it. Thanks.

abc767234318, 2022-07-20:
Thanks a lot !

closed 2022-07-20 by abc767234318