wasm-micro-runtime/wasm-micro-runtime #1299: A stack overflow problem occurred when executing wasm files.

abc767234318, 2022-07-17:
[qjs.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9127206/qjs.zip)

### Steps to Reproduce
```
iwasm qjs.wasm
```
Then I got the following error.
```
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001F28E85BA10
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001F28E90F430
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001F28E8DFF20
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001F28EC0D020
Exception: wasm operand stack overflow
```
And then I do this.
```
iwasm --stack-size=60000  qjs.wasm
```
It seems to work, but I can't type anything through the keyboard.
```
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001A36E232060
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001A36E2A50B0
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001A36E283620
Unhandled exception thrown:  exception code: 0x40010006, exception address: 00007FFE09BE474C, exception information: 000001A36E2BA0A0
QuickJS - Type "\h" for help
qjs > 
```
My execution environment is Windows, but I find that executing on Linux has some problems as well.

wenyongh, 2022-07-18:
@abc767234318 Thanks for reporting the issue! I submitted PR #1300 to fix it, could you please test again?

abc767234318, 2022-07-18:
Thanks a lot.

closed 2022-07-18 by abc767234318