wasm-micro-runtime/wasm-micro-runtime #1304: Fail to load module multiple times ?

Zhangyx24, 2022-07-18:
## Steps to reproduce
Executing the wat file
[test.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9131732/test.txt)
  in C
[test-in-c.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9131733/test-in-c.txt)

Try to load and execute it multiple times.

## Expected behavior
Successfully execute.
And wasmer, wasmtime could repeat the load and execution of the wat file for unlimited times.

## Actual behavior 
WAMR can do it once, but can't in 100times or unlimited times.
[image]

lum1n0us, 2022-07-19:
There are several issues in the test code.

- `wasm_engine_new` and `wasm_store_new` are not pair with `wasm_store_delete` and `wasm_engine_delete`. In wasm-c-api, `wasm_XXX_new` is kind of constructor and `wasm_XXX_delete` is kind of deconstructor. In test code, `wasm_store_t` and `wasm_engine_t` are deleted too many times, and are both invalid in secondary times.

- `wasm_byte_vec_t binary` was destroyed in first loop and become invalid in all next rounds. Still a not paired problem.

- `wasm_byte_vec_t binary` can't be reused. It has to be renewed. In test code, an appropriate way is to `fopen()` and `fclose()` in every loop.

Zhangyx24, 2022-07-19:
Thanks for your answer !

closed 2022-07-19 by Zhangyx24