wasm-micro-runtime/wasm-micro-runtime #1310: Memory access through wasm-c-api

clover2123, 2022-07-20:
Hi
About one year ago, I asked about `wasm-c-api` issue in #677.

It seems like that `wamr` still isolates objects created by wasm-c-api functions.

Following grow API are not supported either.
```
wasm_table_grow
wasm_memory_grow
```

So, current `wamr` does not allow host-side function(wasm-c-api) to directly handle the internal memory.
This approach would be safe in a term of security, but it also restricts users who want to run wasm code through wamr engine.
According to the wasm-c-api list, there exist API functions for deletion like `wasm_memory_delete`.
IMHO this api list means that users could exploit wasm engines freely and it should be users who take charge of any error or defect.

Would you enable it or have any plan for supporting this?

lum1n0us, 2022-07-25:
there is still no plan for growing memories and tables from the host. We will keep you in the loop.

xujuntwt95329, 2022-07-25:
Hi @clover2123, we prefer another approach to implement memory.grow:

You can write an export function with `__builtin_wasm_memory_grow` intrinsic and link with the wasm module, and from the host side you can invoke the exported function to achieve the memory grow, this should be safer than directly expose the `wasm-c-api`

Hope this can help you, thanks~

clover2123, 2022-07-28:
@lum1n0us
@xujuntwt95329 
Thanks! I'll try it with an export function :)