wasm-micro-runtime/wasm-micro-runtime #1317: The implementation of the opendir function appears to be faulty.

abc767234318, 2022-07-21:
```
#include <dirent.h>
#include <stdio.h>
#include <errno.h>

int main(int argc, char **argv) {
  DIR *d;

  char * target = ".";
  if (argc == 2) {
      target = argv[1];
  }
  struct dirent *dir;
  d = opendir(target);
  if (d) {
    while ((dir = readdir(d)) != NULL) {
      printf("%s\n", dir->d_name);
    }

    printf("errno: %d\n", errno);
    closedir(d);
  }

  return(0);
}
```
This is a source code first mentioned in wasmtime issues NO.2493.
My project files are as follows:
[wasmer-2493.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9159586/wasmer-2493.zip)

### Steps to Reproduce
```
iwasm ls.wasm --dir=./testfolder -- testfolder | wc -l
```
The final result is 0. However, the execution result in wasmtime is 203.

wenyongh, 2022-07-23:
@abc767234318 wasmtime supports passing `--dir=xxx` and `--` arguments to wasm app, which are resolved as wasmtime's arguments but not app's, so here command:
`wasmtime test.wasm --dir=./testfolder -- testfolder | wc -l`
is equal to command:
`wasmtime --dir=./testfolder -- test.wasm testfolder | wc -l`

Note that it is a little tricky, not sure what it will occur if the wasm app also accepts `--dir=xxx` and `--` as its arguments. And iwasm doesn't support these usages. The similar command for iwasm should be:
`iwasm --dir=./testfolder test.wasm testfolder | wc -l`

And the result is the same as the result of wasmtime.

abc767234318, 2022-07-23:
@wenyongh Thank you very much for your answer. This is my mistake.

closed 2022-07-23 by abc767234318