wasm-micro-runtime/wasm-micro-runtime #1318: __wasi_fd_filestat_get has an error on Windows.

abc767234318, 2022-07-21:
### Test Case
```
#include <stdio.h>
#include <stdlib.h>
#include <sys/stat.h>
#include <sys/types.h>

#include <wasi/api.h>

int main(int argc, char **argv) {

        __wasi_fdstat_t statbuf;
        printf("__wasi_fd_fdstat_get for stdout: %d\n", __wasi_fd_fdstat_get(fileno(stdout), &statbuf));
        printf("rights for stdout: %llu\n", statbuf.fs_rights_base);
        printf("type for stdout: %d\n", statbuf.fs_filetype);
        printf("has filestat_get rights for stdout: %s\n", (statbuf.fs_rights_base & __WASI_RIGHTS_FD_FILESTAT_GET) != 0 ? "yes" : "no");

        __wasi_filestat_t filestat;
        int filestat_res = __wasi_fd_filestat_get(fileno(stdout), &filestat);
        printf("__wasi_fd_filestat_get for stdout: %d\n", filestat_res);
        if (filestat_res != 0) {
                printf("  E: __wasi_fd_filestat_get returned error: %d\n", filestat_res);
        }

        struct stat sb;
        int f = fstat(fileno(stdout), &sb);
        printf("fstat for stdout (fd: %d): %d\n", fileno(stdout), f);

        if (f < 0) {
                perror("  E: fstat returned error");
        }

        return 0;
}
```
This is a source code first mentioned in wasmtime issues No.2515.
[fstat.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/9159957/fstat.wasm.zip)

### Steps to Reproduce
```
iwasm fstat.wasm
```

### Expected Results
Program execution without error

### Actual Results
```
__wasi_fd_fdstat_get for stdout: 0
rights for stdout: 136314954
type for stdout: 2
has filestat_get rights for stdout: yes
__wasi_fd_filestat_get for stdout: 31
  E: __wasi_fd_filestat_get returned error: 31
fstat for stdout (fd: 1): -1
  E: fstat returned error: Is a directory
```

wenyongh, 2022-07-25:
Hi, the libc wasi implementation in Windows is based on uvwasi library, which doesn't support `wasi_fd_filestat_get` for the stdin/stdout/stderr file descriptors. Other runtimes in Windows should also return error if they use uvwasi. Not sure whether it is meaningful to get the file information of stdin/stdout/stderr?

abc767234318, 2022-07-25:
Thank you very much for your answer.

closed 2022-07-25 by abc767234318