wasm-micro-runtime/wasm-micro-runtime #1323: Massive function call overhead when there are no arguments with wasm_runtime_call_wasm_v()

philhoeb, 2022-07-22:
Hi,

I just did some measurements regarding the overhead when calling WASM functions from native code.
The procedure is as simple as that:
1. Set GPIO
2. Call WASM function which resets the GPIO

Surprisingly wasm_runtime_call_wasm_v() leads to an massive overhead of 6.2 ms in interpreter as well as in AOT mode.
wasm_runtime_call_wasm() (1.3 us) and wasm_runtime_call_wasm_a() (2.7 us) perform way better.

The massive overhead only occurs when there are no arguments to be passed to WASM function. If I add a dummy (int tmp) parameter to my WASM function I measure only about 3.2 us.

Hardware: NUCLEO-H743ZI2
RTOS: Zephyr

Best regards,
Philipp

wenyongh, 2022-07-25:
@philhoeb Thanks for reporting the issue, it seems that it is due to wasm_runtime_call_wasm_a/v allocates memory buffer to store the arguments/results each time. I uploaded PR #1326 to fix it, could you please try again?

philhoeb, 2022-07-25:
@wenyongh PR [#1326](https://github.com/bytecodealliance/wasm-micro-runtime/pull/1326) fixes the issue.
Measurements: (480 MHz; 1000 samples)
| Mode  | wasm_runtime_call_wasm() | wasm_runtime_call_wasm_a() | wasm_runtime_call_wasm_v() |
| --- | --- | --- | --- |
| Interpreter | 1.990 us  | 2.433 us | 3.301 us |
| AOT | 1.306 us | 1.819 us | 2.711 us |

Thank you!

closed 2022-07-26 by philhoeb