wasm-micro-runtime/wasm-micro-runtime #1346: Directories with more than ~150 files are truncated when mapped ?

Zhangyx24, 2022-08-01:
## Steps o reproduce

This is related to https://github.com/bytecodealliance/wasmtime/issues/2493
Compile the C program into wasm : 
[ls.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/9233816/ls.txt)

(1) `iwasm ls.wasm --dir=./testfolder -- testfolder | wc -l`
or 
(2)`wamrc -o ls-wamr.aot ls.wasm` and 
`iwasm --dir=./testfolder ls-wamr.aot -- testfolder | wc -l`

## Expected result
203

## Autual result
0

## Environment
WAMR 05-18-2022
The problem both occur on macOS and ubuntu.

wenyongh, 2022-08-04:
Hi, the issue is like #1317, the argument `--` is forwarded to wasmtime/wasmer runtime, it is handled by them but not wasm app. It is a little tricky and iwasm doesn't support that. The correct command for iwasm should be:
```bash
iwasm ls.wasm --dir=./testfolder testfolder | wc -l
```

Zhangyx24, 2022-08-06:
Thanks a lot!

closed 2022-08-06 by Zhangyx24